dronekit/dronekit-python #19: Change speed in guided mode

mrpollo, 2015-01-08:
@eliao asks if we can support MAV_CMD_DO_CHANGE_SPEED

geeksville, 2015-01-09:
Alas I think this change would be for https://github.com/diydrones/ardupilot.  DroneAPI just passes the cmd through.

eliao, 2015-01-09:
Oh, the ask was to create a droneapi method for creating this command, rather than using message_factory.

bbasso, 2015-01-09:
I thought the API could send any mavlink command, or are implementations for each command required?  Is the following not accurate?

out = api.set(MAVLINK_COMMAND, arg1, arg2, ..., argn)

geeksville, 2015-01-09:
@eliao - oh - in that case, good idea ;-)

geeksville, 2015-01-09:
though btw @eliao - I'm not sure if MAV_CMD_DO_CHANGE_SPEED is supported in any mode other than auto (I haven't checked the code though)

mrpollo, 2015-01-09:
@bbasso we expose helpers like ```takeoff()```, ```goto(<Location>)``` so that you don't have to type the full mavlink command

bbasso, 2015-01-09:
We can changes speed in Guided, and 90% sure it's with that command

bbasso, 2015-01-09:
gottcha--is there a sense in implementing the generic api.set() style command as a catch all for all the helpers that do not have an implementation?

mrpollo, 2015-01-09:
it might be possible passing a dict as a second arg for the different values of attributes in the command.

this is how we handle it right now: https://github.com/diydrones/droneapi-python/blob/master/droneapi/module/api.py#L66-L86

hamishwillee, 2015-06-26:
@bbasso YOu'd be right - the command works fine in GUIDED mode on Copter and is now [documented here](http://python.dronekit.io/guide/copter/guided_mode.html#setting-the-speed).

The problem is that it isn't implemented in guided mode on Plane or Rover (which have much weaker guided mode implementations). 

@mrpollo I think we should provide an API for this and/or #173, but the caveat is that we need to have a strategy for what we do if we're connected to plane or rover (sure, the plane or rover will ignore the command, but it would be better if we could somehow make this obvious to users of the API). If this is useful, we should also work with ArduPilot to get it into the platform (I think it is!)

hamishwillee, 2015-07-21:
Closing this as it possible now in Copter. I created #226 to add the feature as "part of the API".

closed 2015-07-21 by hamishwillee