dronekit/dronekit-python #20: ROI helpers

eliao, 2015-01-12:
It'd be nice to have ROI helpers, rather than using mavlink commands to set them.

bbasso, 2015-01-12:
+1.  It's always been a bit opaque about how ROI persists throughout a mission.  I would imagine a helper could make this more clear.  ROI could be persistent, or just exist till the next command

maruina, 2015-06-03:
I'd like to start working on this.

@hamishwillee @mrpollo do you have some thoughts on that? It's late in Italy and I will look at it more carefully tomorrow, but my first idea is to consider ROI waypoints the same as [Command](http://python.dronekit.io/automodule.html#droneapi.lib.Command)

rmackay9, 2015-06-04:
ROI does persist through a mission by the way in the current official version (AC3.2.1) and higher.

By the way are people here logged into gitter?  https://gitter.im/diydrones/dronekit-python
Some people use it and I think it's the best place for live chat related to dronekit.

hamishwillee, 2015-06-04:
@rmackay9 probably here is not a bad place to discuss specific features though :-) So maybe feasibility check in gitter first, then post here to track activity and get "sign off"

@maruina As Randy says, you can already set a ROI waypoint as a waypoint command in AUTO mode and it will persist through the mission. 

It is also possible to send the MAV_CMD as a custom command in GUIDED mode. However this is a bit ugly though, and I think it might be a very good idea to add a settable Vehicle attribute for this. @mrpollo - what do you think?

Cheers
H

PS If you're feeling free, getting a Home attribute is needed (#105)

hamishwillee, 2015-06-26:
@maruina Are you still working on this? FYI, I [documented the message here](http://python.dronekit.io/guide/copter/guided_mode.html#guided-mode-copter-set-roi). 

An additional point not covered above is that the ROI is cleared on mode change, and that mode in this case also means "movement" command. So if you set ROI and then call Vehicle.commands.goto() the ROI is cleared. If you call Vehicle.commands.goto() and then set ROI it will persist until the mode is changed, you clear it by sending zeros or you change the movement command you use (e.g. moving using SET_POSITION_TARGET_LOCAL_NED]

maruina, 2015-06-26:
I had to put it on standby to work issues. Thanks for the link, I will look at it as soon as possible!

hamishwillee, 2015-07-07:
@maruina Thanks very much. Hope you find time soon. Please add a note here when you start so that someone else can take it over if they find time first!

hamishwillee, 2015-07-08:
@tcr3dr @djnugent This is pretty much a "no brainer" of something we should do. If you have time before @maruina does, please implement :-)