dronekit/dronekit-python #24: Console issues

mattanimation, 2015-03-05:
I've just started working with this api and so far it seems straight forward but I've been having some weird issues that might not be api problems specifically but not sure where else to ask about them. I'm running Yosemite on a macbook pro as fyi. 
The issue is, I will run my script via the USB cable directly ( ex. mavproxy.py --master=/dev/cu.usbmodem1 --cmd="api start spray.py") it works like it should but if I connect via the usb radio adapter (ex. mavproxy.py --master=/dev/cu.usbserial-DN00B8MP --cmd="api start spray.py") then the api hangs anytime I change modes (ex. from GUIDED to ALT_HOLD) and I just see a MAV> console input

geeksville, 2015-03-06:
hmm.  can you try doing it from the mavproxy shell and tell me what you see?  i.e. don't include the api start command.  Instead just type "MANUAL" or "AUTO" at the mavlink shell and see if the mode changes properly.

mattanimation, 2015-03-06:
I connected with the radio usb adapter, here is my terminal output:
admin20081s-MacBook-Pro:example mattmurray$ mavproxy.py --master=/dev/cu.usbserial-DN00B8MP
Logging to mav.tlog
{'sitl': None, 'map': None, 'SOURCE_SYSTEM': 255, 'mission': None, 'TARGET_SYSTEM': 1, 'rtscts': None, 'streamrate': 4, 'console': None, 'auto_protocol': False, 'load_module': [], 'mav09': False, 'speech': False, 'master': ['/dev/cu.usbserial-DN00B8MP'], 'profile': None, 'dialect': 'ardupilotmega', 'baudrate': 115200, 'nowait': False, 'nodtr': False, 'show_errors': False, 'append_log': False, 'aircraft': None, 'continue_mode': False, 'logfile': 'mav.tlog', 'quadcopter': False, 'daemon': None, 'setup': False, 'cmd': None, 'output': [], 'TARGET_COMPONENT': 1}
Loaded module log
Loaded module wp
Loaded module rally
Loaded module fence
Loaded module param
Loaded module relay
Loaded module tuneopt
Loaded module arm
Loaded module mode
Loaded module calibration
Loaded module rc
Loaded module auxopt
Loaded module misc
Loaded module cmdlong
Loaded module battery
Failed to load module: No module named terrain
Loaded module output
Running script /Users/mattmurray/.mavinit.scr
-> module load droneapi.module.api
DroneAPI loaded
Loaded module droneapi.module.api
MAV> AUTO
MAV> MANUAL
Segmentation fault: 11
admin20081s-MacBook-Pro:example mattmurray$

mattanimation, 2015-03-06:
hold on, I think I may have figured it out, I noticed that default baud rate, which has never worked for us in APM, I set the --baud flag to 57600 and it didn't hang that time.

mrpollo, 2015-05-08:
Closing as resolved, if problems persist please create a new Issue.

closed 2015-05-08 by mrpollo