dronekit/dronekit-python #30: AttributeError: 'MPStatus' object has no attribute 'target_system'

igor-nap, 2015-03-30:
Received the "AttributeError: 'MPStatus' object has no attribute 'target_system'" error when attempting to send a mavlink msg. 

In api.py:392, neither self.mpstate.status or self.mpstate.settings had the target_system or target_component attributes set.

Details of setup, error, and .mavinit.scr follow below.

Setup
-------------------------

SITL
$ sim_vehicle.sh --console --map --aircraft test --out=<ip_address>:<port>

MAVProxy
$ mavproxy.py --master=<ip_address>:<port>

Error
-------------------------

Exception in APIThread-0: 'MPStatus' object has no attribute 'target_system'
Traceback (most recent call last):
 File "/usr/local/lib/python2.7/dist-packages/droneapi/module/api.py", line 322, in run
   self.fn()
 File "/usr/local/lib/python2.7/dist-packages/droneapi/module/api.py", line 599, in <lambda>
   APIThread(self, lambda: execfile(args[1], g), args[1])
 File "/home/igor/test_app/main.py", line 13, in <module>
online system 1
AUTO> Mode AUTO
   mgr.Start(api)
 File "/home/igor/test_app/manager.py", line 94, in Start
   self.enableFoo()
 File "/home/igor/test_app/manager.py", line 392, in enableFoo
   self.vehicle.send_mavlink(msg)
 File "/usr/local/lib/python2.7/dist-packages/droneapi/module/api.py", line 244, in send_mavlink
   self.__module.fix_targets(message)
 File "/usr/local/lib/python2.7/dist-packages/droneapi/module/api.py", line 392, in fix_targets
   message.target_system = status.target_system
AttributeError: 'MPStatus' object has no attribute 'target_system'

-------------------------

Contents of my ~/.mavinit.scr

module load droneapi.module.api
module unload terrain
module unload log
module unload battery
module unload fence
module unload rally
module unload calibration
module unload relay
module unload misc
module unload rc
api start /home/igor/test_app/main.py

-------------------------

mrpollo, 2015-04-08:
@igor-nap can you try this again and update your issue with your pip package list with versions ? I'm interested in the mavproxy and droneapi versions you are running

mrpollo, 2015-04-10:
fixed by #53

closed 2015-04-10 by mrpollo