dronekit/dronekit-python #37: The auto-generated API reference is full of open questions

hamishwillee, 2015-03-31:
http://python.dronekit.io/automodule.html

Lots of "FIXME" markup and comments like "This idea is from @rmackay9 - thanks"

These should be closed/removed. If there are some that can be hidden, this can be done using the appropriate sphinx markup (ie you can actually mark sections to build only in a debug build, and in that same build have a table showing the open issues).

hamishwillee, 2015-04-02:
There are easy ways to do this:

1. Use .. only:: draft 
This will add the the stuff in the block only in a "draft" build - omitting it from the final build. 

2. Use the [todo extension](http://sphinx-doc.org/ext/todo.html). I like this because you can mark up any text as a TODO and view it in a todo build (only). In that same build you can use another construct to list all the TODO items.

mrpollo, 2015-04-06:
I agree, lets remove them from Sphinx!

hamishwillee, 2015-04-09:
@mrpollo What release are we up to for the API? Also, how can I work this out for myself?

Specifically in droneapi.lib.Mission and get_mission it has "PLACEHOLDER - will not be included in the release 1 python API, they will only be accessible from the REST API. (This is based on the most likely use-cases wanting a REST interface)"

So if we're up to release 1.x, is this supported?

hamishwillee, 2015-04-09:
I have submitted fix to this in #58:

* All the FIXME comments are still present, but as they appear in a todo directive they are not automatically displayed when you build the documentation. 
* The todos are rendered if you build with the correct flag:
  ```make -e SPHINXOPTS="-D todo_include_todos=1" html```

  (or using our vagrant tool you can do: 
  ```vagrant ssh -c "cd /vagrant/docs && make -e SPHINXOPTS='-D todo_include_todos=1' html" ```

  There is an example [in this temporary dropbox link](https://dl.dropboxusercontent.com/u/3067678/share2/dronekit-python/html/automodule.html). The todos are visible in the text, and are listed at the end. Note that the rendering isn't good - I will raise in a separate defect.

In addition to these changes I made a number of other fixes to the documentation, including linking API reference items, and fixing some errors in names. I have not yet sanity checked the whole API.

I am closing this, as it will be addressed when #58 is merged.

closed 2015-04-09 by hamishwillee