dronekit/dronekit-python #40: First paragraph of Python entry point improvements

hamishwillee, 2015-04-01:
http://python.dronekit.io/ (index.rst)

The text in the first paragraph might be improved:

> DroneKit-Python provides interfaces for creating  applications that run on an onboard Linux computer communicating directly to the autopilot of 3DR-powered vehicles or applications that run on a personal computer communicating with vehicles over a wireless connection.

It does talk about the fact that the API can be used on Linux computer and on ground stations (good). The problems are that it:

* uses term "onboard Linux computer" where we use "Companion Computer" elsewhere. 
* does not specify the ground station OS other than saying "personal computer"
* doesn't cover the benefits of a companion computer (desirable) 
* specifies "autopilot on a 3DR-powered vehicle - and this is suppose to be general.
* The different communication methods for ground station vs companion computer is interesting but not quite as important.

hamishwillee, 2015-04-01:
I've created some text which I believe fixes this and committed it. The text makes it more clear what computing platforms the API runs on and says a little more about the benefits to be had from the API. I've reproduce below. Note that the actual text has links, while this does not:

> DroneKit’s Python API helps you create powerful apps for both ground and air-based control of UAVs.

> The API makes it easy for your ground station apps to get information from UAVs and automate new kinds of flight behaviour. DroneKit-Python code running on a UAV’s Companion Computer can further augment the autopilot — performing tasks that are both computationally intensive and require a low-latency link (e.g. computer vision).

> DroneKit is compatible with all vehicles using the MAVLink protocol (including most vehicles made by 3DR and other members of the DroneCode foundation). The APIs can be used in ground-based person computers running Windows, Linux or Mac OS X and with Companion Computers running Linux.

mrpollo, 2015-04-02:
@hamishwillee is this closed by https://github.com/diydrones/dronekit-python/pull/38 ?

hamishwillee, 2015-04-03:
Yes. I hope you had a quick sanity check of the text - even a tech writer can make mistakes :-)

Note, I had to refresh my cache to view this change. Do we have reasonable server expiry limits on the pages?

Closing this.

closed 2015-04-03 by hamishwillee