dronekit/dronekit-python #43: Provide information and vagrant file on how to contribute to docs?

hamishwillee, 2015-04-03:
We should make it easier for people to get started with contributing to the documents. The good news is that *sphinx_3dr_theme* can now be updated from pip, so it is now possible to set up sphinx and build the source without having to modify any files (previously you needed to modify the conf file in order to replace the theme). 

I've written a vagrant script that does this. The proposal is to put this in /docs/ alongside the .rst files. The (windows) workflow would be:

1. Create a fork of this project
1. Clone the fork locally 
1. Navigate to your docs folder and run "vagrant up". This will bring up the VM, add sphinx and the theme, pull the latest code git, then build (call "make html").
1. Users can then modify the files in their windows file system/notepadd++ and cause a new build by running: ```vagrant ssh  -c "cd /vagrant && make html"```
1. They can commit the changes and do other git activities through windows. (or through the VM).

Is this approach desirable? Is the location of the vagrantfile in docs OK? (I think it is, but just checking).

In addition, I propose to update README.md to add information about how to contribute (specifically to the docs, but more generally it would be useful to cover how people can join the project?).

I would do this by adding a new heading "Contributing" near the end with a link "Contributing to the Documentation". That doc would explain how to use the vagrant file etc. 

OK?

phmagic, 2015-04-03:
Sounds good to me.

mrpollo, 2015-04-03:
@hamishwillee I'm testing your Vagrantfile, I agree with your comments I'll be pushing a branch with your changes so we can reference that, and commit there instead of offline and sharing

mrpollo, 2015-04-03:
I created a pull request #45 that contains the Vagrant changes you propose, please let me know if they make sense, if they do please post your other changes to that branch.

PR #45 (2015-04-07): Vagrant for cross-platform doc builds
@hamishwillee brought up the great idea of adding vagrant so any Windows user can help build our docs as well. #43

I left the Vagrantfile on project root mainly because we need to install the requirements.txt pip packages to be able to build the autodocs, this means your 1 liner needs to change to this:

```bash
vagrant ssh -c "cd /vagrant && make html"
```

@hamishwillee please add to this branch the contributing changes on the README, there are also lots of WARNINGS on the doc builds im not sure how to fix those, can you please help?

fixes #43
closed 2015-04-07 by mrpollo