dronekit/dronekit-python #48: Companion Computer menu link is at wrong level should be submenu)

zlite, 2015-04-06:
That link is should be in the submenu, not a menu header. 

It's now like this:
[image]

It should be as shown here: 
[image]

mrpollo, 2015-04-06:
Agreed, I'll make this change right away.

mrpollo, 2015-04-06:
@hamishwillee regarding your comments on #46 would you be able to submit a PR with a fix for this taking into consideration @zlite comments?

hamishwillee, 2015-04-07:
My comments are now no longer relevant given a "pure-Air" focus for the python docs. So no need. This can probably be closed.

zlite, 2015-04-07:
Now that we've fixed the level of the Companion Computer link, we can remove the second stand-alone top-level link, as shown:
[image]

hamishwillee, 2015-04-10:
This was fixed

closed 2015-04-10 by hamishwillee