dronekit/dronekit-python #54: Simple CherryPy Server

AndrewKe, 2015-04-08:
I can't seem to get a simple CherryPy Server working with DroneKit. The code below works fine when I execute it out of mavproxy using  `python drone.py`  but when I enter mavproxy and execute it using `api start drone.py` , none of the CherryPy debug messages appear and I just get "STABILIZE> APIThread-0 exiting..."

```python
#drone.py

import cherrypy

class HelloWorld(object):
    @cherrypy.expose
    def index(self):
        return "Hello world!"

if __name__ == '__main__':
   cherrypy.quickstart(HelloWorld())
```

I'm trying this on a RaspberryPi2 and the 'drone delivery" demo which also uses CherryPy works on it.

mrpollo, 2015-04-08:
please share a stack-trace and the version of mavproxy, droneapi and cherrypy you are currently running.

Also I wrote an example using CherryPy here https://github.com/diydrones/dronekit-python/blob/master/example/drone_delivery/drone_delivery.py maybe this can help you get started.

AndrewKe, 2015-04-08:
Thank you @mrpollo.

I derived this code from the drone_delivery example and it now works. 
 
```python
import cherrypy

class Drone(object):
    
    @cherrypy.expose
    def index(self):
        return "test"

# Start web server
cherrypy.tree.mount(Drone())
cherrypy.config.update({
                       'server.socket_port': 8080,
                       'server.socket_host': '0.0.0.0',
                       'log.screen': None
                       })
                           
cherrypy.engine.start()
cherrypy.engine.block()
```

mrpollo, 2015-04-08:
What would you say was the part of your code that was causing your errors? (for posterity)

AndrewKe, 2015-04-08:
I think manually starting the web server instead of using cherrypy.quickstart() solved it.

mrpollo, 2015-04-08:
glad we could help, thanks for sharing!

closed 2015-04-15 by mrpollo