dronekit/dronekit-python #59: Should the genindex be in the TOC?

hamishwillee, 2015-04-09:
This file is autogenerated, but not linked from anywhere http://python.dronekit.io/genindex.html

How useful is it? Should it be in the sidebar?

mrpollo, 2015-04-09:
Probably below API reference? do you find it useful?

hamishwillee, 2015-04-10:
Not "super" useful - and this will be less useful when our class documentation CSS is improved and it is possible to read it more easily. 

If we do want this in the TOC it has to be added as an explicit URL link to the index as shown below (ie you can't link to the version just built!):
```
Index <http://python.dronekit.io/genindex.html>
```
We could however just add it as a link on a home page or similar and then it can be linked "properly" using :ref:`genindex`.

I propose we do not add this. If neither of us are convinced it has clear value, then probably it is just clutter?

mrpollo, 2015-04-10:
If you can add it as a submenu item of the api reference I vote to add it, if not lets leave it out

hamishwillee, 2015-04-13:
I can, included in pull request #58 

Note, the included doc is hardcoded to point to the current python version of the file - so if you build this locally the link will take you to the main site. This is not desirable, but there is no way around it (as discussed above - you cannot include genindex in a TOC except by providing a full URL).

Closing, as this will be delivered by the pull request

mrpollo, 2015-04-13:
thanks @hamishwillee