dronekit/dronekit-python #61: Attribute observer callbacks pass in the name of the attribute, not the value

eliao, 2015-04-09:
This seems unexpected to me.  

eg, If I call:
vehicle.add_attribute_observer('mode', mode_callback)

mode_callback is called with 'mode' as a parameter.  I would expect to be passed the value of vehicle.mode.  I'm not sure why I would want the name of the attribute, as I already know the name.  

Perhaps the intention is for a single callback to handle multiple attributes?  That seems odd to me, and the API does not enforce such a construct.

mrpollo, 2015-04-09:
I agree, this is confusing, right now the right way of doing reading the value is just reading through the global on the callback listener, the value will be updated before the trigger of callbacks relevant source for this behavior is here:

**APIModule**
https://github.com/diydrones/dronekit-python/blob/master/droneapi/module/api.py#L395-L397

**HasObservers**
https://github.com/diydrones/dronekit-python/blob/master/droneapi/lib/__init__.py#L179-L193

I'm not entirely sure why we went this route, but I'll find out and maybe try to add a feature request for next version to also include param values, what would you expect the normal behavior form a callback wold be?

eliao, 2015-04-09:
I would expect the callback to pass in the new value of the attribute.  

Preferably only when it changes, as referenced by https://github.com/diydrones/dronekit-python/issues/60

tcr3dr, 2015-11-09:
This is changed as of #426 to have the value passed in.

closed 2015-11-09 by tcr3dr