dronekit/dronekit-python #63: Adding Camera support for Companion Computers

robnsngh, 2015-04-14:
USB camera and other cameras support for companion computers. Potentially allowing developer to access live stream over LTE from the drone. Roadmap to create "periscope" like app for drones.

mrpollo, 2015-04-14:
@robnsngh I think this is of value to our docs, I will make sure we have some "supported" cameras or at least a list of what we have tested in there.

LTE Live stream, I think that we are going to give some level of "support" to achieve streaming through the API, or at least set a standard of "recommended" settings, but might not go as far as creating a "periscope" like app for drones, but that is the type of apps we would love to see come out of the community, and we are committed to help create that or any other type of app that uses DroneKit

@hamishwillee do you think we should add a supported cameras on our companion computer section of the docs?

hamishwillee, 2015-04-14:
@mrpollo I would split this into two parts - a documentation issue, and an API-enhancement issue. What they are mostly asking for here is the later - support in the API for control of video, including an endpoint in the API to get video frames. This is not AFAIK supported in ArduPilot yet, but as Solo has this we could define an API for getting the frames and some way of telling if the computer supports a camera or not.

The rest of my response answers the "documentation" part of the query. The simple answer is "no", but we should have a "Using the camera how to".

-----

So ArduPilot supports integration with a gimbal for pitch/roll/yaw and for the camera itself all we have currently is the camera shutter trigger. This trigger comes off a servo (PWM) or relay. The actual cameras have millions of different types of remote control mechanisms from IR to cable (with heaps of different connectors and protocols). As a result, between the camera and the autpilot output we need to put something which translates the servo/relay output to the form understood by the camera. As an example, for some canon cameras you can use [CHDK](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-cameras-and-gimbals/common-apm-to-chdk-camera-link-tutorial/)

From the perspective of DroneKit we support the shutter triggering and gimbal control through a waypoint [Command](http://python.dronekit.io/automodule.html#droneapi.lib.Command) - we send a command to set gimbal position and to trigger the shutter at certain points (or at certain spacings based on airspeed).

IN summary:

1. Essentially we already support USB and other cameras for stutter triggering. However at the dronekit layer this is irrelevant. 
2. We don't yet have standardised support in ArduPilot what people would quite like - video triggering, sending video packets to companion computer through IP interface. Adding a standard interface for doing this to ardupilot and dronekit is desirable.

Given all that my answer is "no". We should not add a supported cameras section. 

What we should do is add a guide topic for important use cases that explains what you can and cannot do, with links and examples. In this case it would be called: "Using the camera". We would provide an overview like above with links to ardupilot. We would need to show how to create a waypoint to trigger the camera using the API. We should also show how to detect if camera has been installed/configured through the API. Have we any examples of that?

Make sense?

mrpollo, 2015-04-14:
I mostly disagree with you @hamishwillee, while I agree that we already have some level of support for CHDK and triggers via waypoints I don't think this is the solution going forward, and the autopilot shouldn't be involved in any video transaction since video is way lower priority.

Yes we need to include support for cameras on DroneKit, and no, it's not irrelevant, Video is a pending problem in the eco-system, if DroneKit helps providing a standard on software, protocols, encoders and hardware, we will be guiding and solving this problem for most users (at least DroneKit users).

What is still pending in video land on the drone community for what I can see is:

- Streaming of digital video from vehicle to ground
- Computer Vision software stack
- Uniform camera control, a library that knows how to talk to the camera regardless of hardware (see notes below)
- Documentation, which software, what hardware, what works better, what is open, what is closed (mostly codecs)

This came up at ELC, we all use video now, but everyone is still trying out ideas and doing trial and error to work out what works better for them, the community already invested in this space, what DroneKit has right now is an opportunity to not just be a software API, but also help push standards, how and why?, we write libraries to help unify this inside DroneKit, we provide documentation, there are many cameras/gimbals out there, I am not trying to reinvent the wheel here, but if we define what is better and what works best, which is what I think @robnsngh is trying to get from us here, then we push things forward, of course this is only my opinion, I'm sure @phmagic has more to say on this, and not necessarily agree with me.

====
I think your guide idea is great, and something we can do right now, yes we have some camera examples, lets discuss somewhere else about writing this guide.

hamishwillee, 2015-04-15:
Hi @mrpollo 

As a documenter, my response is mostly about "what support we offer now". I think it makes sense to provide a clear story on that for our users. 

So given that, what don't you agree with about the proposed documentation solution?

(and yes, your points regarding our future direction make a lot of sense. The reasons I want to separate the documentation and future-direction problem is because I'm not the right person to comment on that :-)

hamishwillee, 2015-04-15:
@mrpollo  - ah, didn't read your message properly! "I think your guide idea is great, and something we can do right now, yes we have some camera examples, lets discuss somewhere else about writing this guide."

OK, we agree on that. Yes, lets discuss offline.

silviorevelli, 2015-04-24:
Video, gimbal, lidar is not an autopilot's affair. Such low-priority task should be addressed by companion computer. FYI: 4G video streaming has been already achieved www.4gmetry.com

ricardotk, 2015-11-18:
Any updates on this? At least about the guide idea