dronekit/dronekit-python #65: Hide non-implemented APIs from documentation

hamishwillee, 2015-04-17:
The API contains a number of APIs that are not fully implemented, and may never be. We do not intend people to use them at the current time. They are however publically visible and currently documented.

* class Mission
* get_mission
* ConnectionInfo
* web_connect

Obviously it is best that these are either fully implemented and so are usuable or are removed from the public API. In the meantime I propose:

* Hide in the documented API. We can do this by adding this flag to the rst automodule directive

```
   :exclude-members: Mission, get_mission, ConnectionInfo, web_connect
```

* Remove all other mention of them in the documentation
* Include status warnings ("not supported") in the source so that people reading the source can see these are not public and are intentionally not documented.

Note: This does not apply to "deprecated" APIs. APIs that might have been used by developers in the past need to be tagged as deprecated.

Thoughts?

mrpollo, 2015-05-04:
first list is ok, I'm going to have to look at ```notify_observers`` and ```vehicle.delete```

hamishwillee, 2015-05-05:
As above, have done the first part (Mission, get_mission, ConnectionInfo, web_connect, AuthInfo) in the pull request.

hamishwillee, 2015-05-06:
As discussed, Vehicle.delete is internal and notify_observers remains public. 

This item is closed by #80

closed 2015-05-06 by hamishwillee