dronekit/dronekit-python #66: Flight Replay example uses incorrect API

hamishwillee, 2015-04-21:
The [Flight Replay example](http://python.dronekit.io/example_3.html#setting-the-new-waypoints) sets waypoints using mavutil.mavlink.MAVLink_mission_item_message directly. As I understand the API the correct type for a command in this API is "Command" (which is not even imported)

```
    cmd = mavutil.mavlink.MAVLink_mission_item_message(0,
                                                     0,
                                                     0,
                                                     mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT,
                                                     mavutil.mavlink.MAV_CMD_NAV_WAYPOINT,
                                                     0, 0, 0, 0, 0, 0,
                                                     lat, lon, altitude)
    cmds.add(cmd)
```

To fix this I believe the change is trivial, because the Command is just a thin wrapper around MAVLink_mission_item_message

```

from droneapi.lib import VehicleMode, Command
...
    cmd = Command(0,
                                                     0,
                                                     0,
                                                     mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT,
                                                     mavutil.mavlink.MAV_CMD_NAV_WAYPOINT,
                                                     0, 0, 0, 0, 0, 0,
                                                     lat, lon, altitude)
    cmds.add(cmd)
````

mrpollo, 2015-04-21:
@hamishwillee did you try this in SITL?

hamishwillee, 2015-04-21:
I ran both the original code and with my changes above. In both cases the output "looked reasonable" and was the same. 

BUT, I did NOT go through the code fully yet and confirm that it is all working as expected. So "looking reasonable" is not a canonical "this code works perfectly".

hamishwillee, 2015-05-06:
Tried this in SITL and the fixed file works. I have made this change. Closing as fix is in #81

PR #81 (2015-05-06): Fix flight replay source code and docs to use the Command class
This is to fix #66 . It also includes some minor typographic issues in the documentation.
closed 2015-05-06 by hamishwillee