dronekit/dronekit-python #70: Add CommandSequence.takeoff to the public API

hamishwillee, 2015-04-28:
This is defined in the implementation class (not the interface class) and does not appear in the documentation. 

It needs to be made visible in the documentation, ideally by defining it in the interface.

Note: I split this out from #64 so I can assign ownership to @mrpollo .

hamishwillee, 2015-05-22:
#134 adds the method to the documentation as part of the "class docs". I'm leaving this open because even though this will fix the documentation it doesn't do it in the right way - which is to add the method into the __init__.py .

tcr3dr, 2015-11-09:
@hamishwillee Is this technically a bug still? Looks like it's part of CommandSequence now (although we may move that into Vehicle proper?)

hamishwillee, 2015-11-09:
@tcr3dr No, this can be closed. 

I will move the documentation for all methods under the methods themselves (out of the "class level" docs) as soon as we get some API stability.

closed 2015-11-09 by hamishwillee

tcr3dr, 2015-11-09:
@hamishwillee Sounds great.