dronekit/dronekit-python #72: Move channel_override examples out of simple_goto and small_demo

hamishwillee, 2015-04-29:
According to @tridge overriding channels is intended for simulating user input (ie in autotest) or for implementing joystick behaviour. It should not be used to directly drive a vehicle, except as a last resort. Instead the appropriate mavlink commands should be used.

The following examples set the channel_override in  a way that we would not recommend:

* https://github.com/diydrones/dronekit-python/blob/master/example/small_demo/small_demo.py#L80
> https://github.com/diydrones/dronekit-python/blob/master/example/simple_goto/simple_goto.py#L26

I propose that we:

* strip this from small_demo and simple_goto, and create a small independent demo that shows just this behaviour. 
* add a warning in the new example's code and docs about this use case
* replace the throttle setting in the simple_goto with a more appropriate command for setting the throttle.

hamishwillee, 2015-05-21:
Pull #124 will fix the small_demo.py case. I have left the RC override stuff there, but it is very clearly marked as "do not use".

tcr3dr, 2015-07-16:
It appears that simple_goto no longer includes RC code.

Should we still split out RC code from vehicle_state? I would be in favor. It seems like it's an "advanced" topic and we want to not encourage that in the basic example, no matter how many warnings are attached.

hamishwillee, 2015-07-17:
@tcr3dr I thought about this a lot and decided that it is more associated with vehicle state than anything else, and that creating an extra "advanced use" example to support this would actually highlight it more than I wanted. I do see your argument though - if you feel strongly add a final confirmation and I'll create another very simple example for "setting rc overrides". Otherwise just close this.

tcr3dr, 2015-07-28:
I'd like to propose separating this then. Considering some of the long-term goals of maintaining abstractions over MAVLink and RC input, and since vehicle_state.py will eventually accumulate more features we want to highlight (and are more important), moving out "advanced" features gives us that level of flexibility. But this is totally low-priority.

hamishwillee, 2015-07-28:
@tcr3dr OK. I am leaning this way too.

tcr3dr, 2015-11-07:
I believe this is done.

closed 2015-11-07 by tcr3dr

hamishwillee, 2015-11-07:
Yes, this was done ages ago :-)