dronekit/dronekit-python #74: Attribute observers result in endless output, even when script exited

hamishwillee, 2015-04-29:
Remove the comment [on this line in small_demo.py](https://github.com/diydrones/dronekit-python/blob/master/example/small_demo/small_demo.py#L70) to re-enable the mode.

So the code has
```
def mode_callback(attribute):
    print "Mode changed: ", v.mode

v.add_attribute_observer('mode', mode_callback)
```

When executed, the code runs. After the thread exits the mavproxy command prompt prints out the mode forever as shown below. This is a bug of some sort right?
```
AUTO>
AUTO> module load droneapi.module.api
AUTO> api start small_demo.py
AUTO> Mode: VehicleMode:AUTO
Location: Location:lat=-35.3632603,lon=149.1652287,alt=0.0,is_relative=False
Attitude: Attitude:pitch=0.00512621784583,yaw=-0.0500796176493,roll=0.00491848401725
Velocity: [-0.08, 0.08, 0.0]
GPS: GPSInfo:fix=3,num_sat=10
Armed: False
groundspeed: 0.0
airspeed: 0.0
Requesting 95 waypoints t=Wed Apr 29 14:56:15 2015 now=Wed Apr 29 14:56:15 2015
Home WP: MISSION_ITEM {target_system : 255, target_component : 0, seq : 0, frame : 0, command : 16, current : 0, autocontinue : 1, param1 : 0.0, param2 : 0.0, param3 : 0.0, param4 : 0.0, x : -35.3632621765, y : 149.165237427, z : 584.090026855}
Current dest: 1
Disarming...
Arming...
Overriding a RC channel
Current overrides are: {'1': 900, '4': 1000}
RC readback: {'1': 1500, '3': 1000, '2': 1500, '5': 1800, '4': 1500, '7': 1000, '6': 1000, '8': Got MAVLink msg: COMMAND_ACK {command : 400, result : 0}
1800}
Cancelling override
APM: Arm: Mode not armable
APIThread-0 exiting...
Got MAVLink msg: COMMAND_ACK {command : 400, result : 4}
Got MAVLink msg: COMMAND_ACK {command : 11, result : 0}
Mode changed X:  VehicleMode:AUTO
Mode changed X:  VehicleMode:AUTO
Mode changed X:  VehicleMode:AUTO
Mode changed X:  VehicleMode:AUTO
Mode changed X:  VehicleMode:AUTO
Mode changed X:  VehicleMode:AUTO
Mode changed X:  VehicleMode:AUTO
Mode changed X:  VehicleMode:AUTO
Mode changed X:  VehicleMode:AUTO
Mode changed X:  VehicleMode:AUTO
Mode changed X:  VehicleMode:AUTO
...
```

mrpollo, 2015-05-06:
I need to confirm if this is a MAVProxy problem or DroneKit

tcr3dr, 2015-07-29:
Fixed by #255.

closed 2015-07-29 by tcr3dr