dronekit/dronekit-python #77: DOCs: API Reference is building an old version (somehow)

hamishwillee, 2015-05-06:
The API Reference on Python IO does not reflect the current API. 

The documentation is auto-generated from [__init__.py](https://github.com/diydrones/dronekit-python/blob/master/droneapi/lib/__init__.py). If you look at this source you can see that this differs quite a lot from [what is published](http://python.dronekit.io/automodule.html). When I build this locally the built reference matches the file ([example here on my dropbox](https://dl.dropboxusercontent.com/u/3067678/share2/dronekit-python/html/automodule.html) ).

This does not appear to be a caching issue on the page or with the "whole build" (since the rest of the docs on python.dronekit.io are correct and reflect changes from the latest pull request). 

@mrpollo I think this is something very wrong with Sphinx build. Possibly it has failed to build the automodule? In any case, this is important.

mrpollo, 2015-05-07:
the docs server wasn't updating the reference to the dronekit-python package it was using to generate the autmodule, this has been corrected and the latest changes are now on http://python.dronekit.io/automodule.html

closed 2015-05-07 by mrpollo

hamishwillee, 2015-05-07:
Perfect thanks - good to see all that doc work gone live!