dronekit/dronekit-python #78: Vehicle parameter update observer not called, possibly due to timeout

hamishwillee, 2015-05-06:
I am try to observe a change made to a Vehicle.parameter value. The test code at the bottom of this comment registers an observer on THR_MIN. It then reads the value and attempts to change it. The code then writes a different value, sleeps 5 seconds, then prints it again.

In all cases, the console reports a timeout setting the parameter. However when the value is next printed it matches the new setting. At no time is the parameter observer called:
```
api start /vagrant/small_demo.py
AUTO> Param THR_MIN: 200.0
timeout setting THR_MIN to 50.000000
Param: 50.0
```

@mrpollo Please tell me how to extend the timeout so I can test the parameter observer (and extend it in the API if that is what your investigations reveal)

```
import time

# First get an instance of the API endpoint
api = local_connect()
# get our vehicle - when running with mavproxy it only knows about one vehicle (for now)
v = api.get_vehicles()[0]

def thin_callback(attribute):
    print "CALLBACK changed: ", v.parameters['THR_MIN']
	
v.add_attribute_observer('THR_MIN', thin_callback)

# Print the value of the THR_MIN parameter.
print "Param THR_MIN: %s" % v.parameters['THR_MIN']

# Change the parameter value to something different.
v.parameters['THR_MIN']=50
v.flush()

# Print the value of the THR_MIN parameter a bit later
print "Param: %s" % v.parameters['THR_MIN']

# Change the parameter value to something different.
v.parameters['THR_MIN']=100
print "sleeping ..."
time.sleep(5)
print "awake ..."
v.flush()
print "Param: %s" % v.parameters['THR_MIN']

# Change the parameter value to something different.
v.parameters['THR_MIN']=200
v.flush()
print "Param: %s" % v.parameters['THR_MIN']
```

mrpollo, 2015-05-07:
the callback might have never registered since ```THR_MIN``` isn't an attribute of ```Vehicle```, only direct attributes of ```Vehicle``` are being called, in this case you would need to observe ```parameters``` for changes instead

closed 2015-05-07 by mrpollo

hamishwillee, 2015-05-07:
@mrpollo Observing Vehicles.parameters is pointless since even if it worked and the callback was run, you'd have to query every single parameter to find out what the current values are (and you still wouldn't know what changed). The docs say you can directly observe Parameters and that makes sense because it implements HasObservers. I've asked the question more directly in #89