dronekit/dronekit-python #85: message_factory and send_mavlink(message) docs have incorrect advice

hamishwillee, 2015-05-06:
``message_factory`` and ``send_mavlink(message)`` state that custom commands are a catch-all/back door in case we miss something in the API. 

This is incorrect. These are also useful for sending arbitrary commands in guided mode. 

----

In addition, we need guide documentation on these. The documentation should cover that there are a number of ways to send messages. For example COMMAND LONG is a basic message format that can be used to send any seven parameter command. So you could use:
```
vehicle.message_factory.command_long_encode to send a MAV_CMD_CONDITION_YAW 
```

However you could also send this command using its own method like below. The benefit being that you don't have to think about command ids, sequence numbers etc.:

```
vehicle.message_factory.mav_cmd_condition_yaw_encode
```

mrpollo, 2015-05-07:
:+1: agreed, you are correct

hamishwillee, 2015-05-07:
See #90 for updates to the API Reference to explain that send_mavlink is useful for sending arbitrary messages, particularly in GUIDED mode. I will leave this open until I create a "Custom commands" page.

closed 2015-05-08 by mrpollo

jsjain007, 2016-06-06:
yes you are perfectly correct, now it makes more sense of the documentation
(y)

jsjain007, 2016-06-06:
:+1:  now it makes more clear how exactly to send commands, docs sure needs to be updated with more examples on sending mavlink messages.

hamishwillee, 2016-06-06:
@jsjain007 Please don't hang off closed issues - if you have a genuine issue then add the error as a separate case. This is not actually an issue - there are enough examples on the topic, you just have another error. The right place to raise that is the discussion boards: https://discuss.dronekit.io/c/python

I suspect your problem is using an older version of pymavlink which does not include the takeoff command.

jsjain007, 2016-06-07:
sure

FFFgrid, 2019-11-25:
May I ask a queston?
When I run 'vehicle.message_factory.mav_cmd_condition_yaw_encode',it shows that vehicle does not have an attribute named message_factory,Why?

PR #90 (2015-05-08): Fixes APIRef around GPSInfo and send_mavlink
Fixes #85 and part of #86.