dronekit/dronekit-python #89: How do you register/observe changes to a particular Vehicle parameter

hamishwillee, 2015-05-07:
class Parameters implements HasObservers interface, and it should therefore be possible to register to observe parameter changes (this is also indicated by the original documents). 

I tried this, but while the code will run, the observer is not called (possibly because THR_MIN is not an attribute):
```
v.add_attribute_observer('THR_MIN', thin_callback)
```
I tried this, but while the fails with some kind of not-supported error
```
v.parameters.add_attribute_observer('THR_MIN', thin_callback)
```

How is this done?

geeksville, 2015-05-12:
This is meant to be supported and if it doesn't work it would be easy to add.

hamishwillee, 2015-05-12:
Thanks @geeksville

@mrpollo I think "we" should add this. The question is, will it be done in a timely manner, or should I update the documentation now to reflect that it is not possible right now?

mrpollo, 2015-05-13:
update docs to reflect current state of API and please create a new issue assigned to me to add this feature

@geeksville thanks!

hamishwillee, 2015-05-13:
OK, #107 is the issue for the missing functionality. Am assigning this to me to fix the problem.

hamishwillee, 2015-05-13:
Documentation fixed by #108. Closing.

closed 2015-05-13 by hamishwillee