dronekit/dronekit-python #93: Move/tidy up /example/documentation

hamishwillee, 2015-05-09:
The example documentation is currently in two places:

* https://github.com/diydrones/dronekit-python/tree/master/example/documentation
* https://github.com/diydrones/dronekit-python/tree/master/docs/examples

Action is to move the documentation all into the official docs. First step to simply remove the content of migrated docs and add links to where the content is now. When all examples are migrated, then remove the stubs and perhaps just have a readme in root of /master/example/ explaining where the docs are.

@mrpollo - make sense?

mrpollo, 2015-05-09:
Agreed.

hamishwillee, 2015-05-11:
This is closed by #96. Note that there are undocumented examples still (e.g. client_sketch.py). I will raise separate issues for those.

closed 2015-05-11 by hamishwillee