dronekit/dronekit-python #95: Loading DroneAPI module fails on RasPi

zlite, 2015-05-11:
Regarding: http://python.dronekit.io/examples/running_examples.html

Using RasPi

Although I followed the direction and Git Cloned the DroneAPI code (git clone http://github.com/diydrones/droneapi-python.git) and changed directory to the example folder (cd droneapi-python/example/small_demo) and successfully started MAVproxy, I still got a missing file error when I entered: "module load droneapi.module.api"

HOLD> Received 374 parameters
module load droneapi.module.api
HOLD> Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 260, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 390, in import_package
    mod = __import__(name)
ImportError: No module named mavproxy_droneapi.module.api

Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 260, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 390, in import_package
    mod = __import__(name)
ImportError: No module named droneapi.module.api

hamishwillee, 2015-06-26:
@zlite The instructions changed over time, and now state "To run the examples you will need to set up DroneKit as described in Getting Started. Once you have the source code, examples are started as described in the Getting Started section Running an app/example."

So is this working if you follow the instructions in Getting Started?

Serveurperso, 2015-07-10:
Same problem here,

Connect tcp:192.168.43.111:5762 source_system=255
Logging to mav.tlog
Running script C:\Users\Pascal\AppData\Local\MAVProxy\mavinit.scr
-> set moddebug 2
-> module load graph
Loaded module graph
-> module load console
Loaded module console
Traceback (most recent call last):
  File "D:\A\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\Scripts\mavproxy.py", line 264, in load_module
    m = import_package(modpath)
  File "D:\A\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\Scripts\mavproxy.py", line 394, in import_package
    mod = __import__(name)
ImportError: No module named mavproxy_droneapi.module.api

DroneAPI loaded
Loaded module droneapi.module.api
MAV>

hamishwillee, 2015-07-11:
It is irritating, but I don't consider "urgent" because as you will note the library is actually loaded. I'll modify the issue title appropriately and try to debug next week.

kcarring, 2015-08-02:
I'm getting the same result after doing the Getting Started setup twice (and a friend did it once on a different computer and got the same result), so the api start commands don't work us.
What are these modules and where is it looking for them?

hamishwillee, 2015-08-02:
@kcarring - So if you're on mavproxy prompt and you enter ``module load droneapi.module.api`` what do you see?
If you look at the console log [above](https://github.com/dronekit/dronekit-python/issues/95#issuecomment-120409870) the module is being loaded in the end, so I want to confirm you have different symptoms.

kcarring, 2015-08-02:
Mine are similar to zlite's symptoms:

STABILIZE> module load droneapi.module.api
Traceback <most recent call last>:
  File "<string>", line 265, in load_module
  File "<string>", line 395, in import_package
ImportError: No module named mavproxy_droneapi.module.api

Traceback <most recent call last>:
  File "<string>", line 265, in load_module
  File "<string>", line 395, in import_package
ImportError: No module named droneapi.module.api

Failed to load module: No module named droneapi.module.api.

Then it suggests 'set moddebug 3'...when I do that I get the exact same result.

hamishwillee, 2015-08-03:
@kcarring Well, that means what it says - the module can't be loaded. That usually means it isn't in a path where it can be found by MAVProxy.

Can you post your whole build log in a gist for review - first this:

```
sudo apt-get install python-pip python-dev python-numpy python-opencv python-serial python-pyparsing python-wxgtk2.8
```

Then this: 
```
sudo pip install droneapi upgrade
```

kcarring, 2015-08-03:
I'm using Cygwin on Windows. Do you mean the mav.tlog file?
When I installed, I started the WinPython Command Prompt as Admin and typed:
pip uninstall python-dateutil
pip install droneapi

When I try to type pip install droneapi upgrade
it says that it can't find any upgrade downloads/distributions...

Is the droneapi.module.api a file somewhere?

hamishwillee, 2015-08-03:
@kcarring 

Please confirm:

- you are using a SITL vehicle running on Windows in Cygwin
- you are connecting to this on the same machine using a WinPython command prompt?

> Is the droneapi.module.api a file somewhere?

It is a set of files that should have been installed when you did

```
sudo pip install droneapi
```
If you do ``pip list`` is the droneapi component shown?

Right now I am trying to write an updated guide that uses a dronekit windows installer. Might be best of to wait until the end of the day and I can attach a link to the docs - you could use that instead?

kcarring, 2015-08-03:
Yes I was able to get the SITL setup and compiled on a Windows machine in Cygwin using:
make sitl -j4
I'm not sure what you mean on  the 2nd question. I start SITL from Cygwin via:
sim_vehicle.sh -j4 --map
The 3 windows come up, the map shows the drone and I can type commands and watch the simulated drone move.

When I type pip list (in WinPython) it does show droneapi (1.4.2)

Thanks much! I'll wait for the new windows installer.

hamishwillee, 2015-08-03:
@kcarring Great, I'm working on the new docs now. YOu can track from #164.

Setting up DK has two parts - setting up the vehicle (if simulated), and setting up dronekit to talk to it. My second question above was confirming that you're using WinPython for the DroneKit environment. This is important, because you could, for example, decide that you'll not bother with setting up a separate dronekit environment, but run from the mavproxy terminal set up with Cygwin/SITL. THis is possible and reasonable if you're only testing in SITL.

hamishwillee, 2015-08-03:
@kcarring Please see modified getting started in #268 using the windows installer.

Installation is much simpler on windows. The downside is that behaviour is slightly different for making the connection (done after starting MAVProxy) and for running examples (you need to specify the full path to examples because you don't have the ability to start MAVProxy within a specific directory. 

Feedback on this welcome.

kcarring, 2015-08-03:
I got it to work! Thank you! (But  when I type another command into the MAVProxy window the output from the previous command disappears. Also the output of the vehicle_state.py occasionally seems to overwrite parts of itself.)
Here's what I did to get it working though:
Installed a new MAVProxy and dronekit-windows via running:
http://firmware.diydrones.com/Tools/MAVProxy/MAVProxySetup-latest.exe
and
https://s3.amazonaws.com/dronekit-installers/windows/dronekit-windows-latest.exe
Then I proceeded as before starting everything from Cygwin in the ardupilot/ArduCopter directory that has my built ArduCopter.elf where I typed:
sim_vehicle.sh -j4 –map
Then (I had already copied the example .py files into the ardupilot/ArduCopter directory) in the mavproxy.exe window I typed:
module load droneapi.module.api
api start vehicle_state.py

(On the module load I did get the same error that Serveurperso had, but the api start command now works for me.)

hamishwillee, 2015-08-04:
@kcarring Glad you got it working. I hope to update the SITL instructions so you will no longer need cygwin either. That will remove some confusion, because the standard Windows instructions mean that you end up with multiple consoles and mavproxy windows floating around.

tcr3dr, 2015-11-07:
I'm going to close this issue, as we are no longer using MAVProxy.

closed 2015-11-07 by tcr3dr