dronekit/dronekit-python #100: Vehicle.waypoint_home not implemented

hamishwillee, 2015-05-12:
``Vehicle.waypoint_home`` was in the original documentation, but is not present in the implementation (MPVehicle). So if you try the following you will get an exception.

```
print "waypoint_home: %s" % v.waypoint_home
```
First action here for me is to remove the variable from the documentation as it is "not implemented". 

@mrpollo it sounds like this would be a useful thing to have. Do you know whether this is something that will be added to the API? If so I should assign this back to you to implement?

hamishwillee, 2015-05-12:
Research indicates that implementing this API this probably isn't possible currently. It is possible to [set the home position in a mission](http://planner.ardupilot.com/wiki/common-planning-a-mission-with-waypoints-and-events/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_do_set_home) but it is not 100% clear whether this is the same "variable" as is used in other modes. It is also not obvious that there is any way to get the home position. There was a brief discussion on this in the [discussion group some years ago](https://groups.google.com/forum/#!topic/mavlink/SL0oTUJx1fE).

geeksville, 2015-05-12:
definitely possible, it was probably inadvertently lost when full command access was added..  vehicle.commands[0] is guaranteed to be home.  It would be easy to add it back.

hamishwillee, 2015-05-12:
@geeksville Thanks very much. 

@mrpollo - do you want to add it back or should we just document vehicle.commands[0] gets the right value? My preference is to add it back to make it more visible. 

While we're at it, can/should we make this have a setter as well? That makes sense to me because then you could easily change the home position outside of missions.

hamishwillee, 2015-05-13:
Closed by #108

closed 2015-05-13 by hamishwillee

maruina, 2015-06-04:
I have a couple of doubts:  
* What should be the type of waypoint_home? A Location instance? An array with lat, lon, alt?
* Let's say that I set a waypoint_home and then I create a mission: should my home overwrite vehicle.commands[0] or vice versa? Or should be they always kept separate from each other? Should my waypoint_home be None unless I set a mission?

hamishwillee, 2015-06-05:
@maruina All good questions. I would say

> What should be the type of waypoint_home? A Location instance? An array with lat, lon, alt?

Yes, a Location instance.

> Let's say that I set a waypoint_home and then I create a mission: should my home overwrite vehicle.commands[0] or vice versa? Or should be they always kept separate from each other? Should my waypoint_home be None unless I set a mission?

The home position is (currently) stored in the 0th waypoint, and is automatically set by the autopilot, usually when it first gets GPS lock (though sometimes it is set at takeoff or when arming occurs - not sure). I THINK that you can't upload your own value for this, but that is worth checking.

So I recommend doing some testing. If you can't set the value, then just make it a read-only attribute. If you can set it, then do (but we'll need to make it clear that the value may be over-written at any time by the autopilot).

Does that make sense?