dronekit/dronekit-python #110: Vehicle notifier should only allow a callback to be added/removed once for a particular attribute

hamishwillee, 2015-05-13:
``Vehicle.add_attribute_observer`` will add an observer as many times as it is called, and  ``Vehicle.remove_attribute_observer`` will remove as many instances of the observer as times it is called.

IMO you might want to assign multiple observers to the same attribute, but I would expect them to be different. You might in theory want to also apply the same observer to multiple attributes. However a callback should really only be added once as an observer for an attribute, however many times it is called to be added.

The code below shows that if add_attribute_observer is called 4 times to add an observer, and then removed once, the callback is called 3 times. It should be called once, however many times it is added.

```
# Now change the vehicle into STABILIZE mode
v.mode = VehicleMode("STABILIZE")
	

# Define example callback for mode
def armed_callback(attribute):
    print "Armed-state changed: ", v.armed

v.add_attribute_observer('armed', armed_callback)
v.add_attribute_observer('armed', armed_callback)
v.add_attribute_observer('armed', armed_callback)
v.add_attribute_observer('armed', armed_callback)
v.remove_attribute_observer('armed', armed_callback)

print "Disarming..."
v.armed = False
v.flush()
```

hamishwillee, 2015-07-21:
This will be fixed by #224. Check docs at that point to see if update required.

tcr3dr, 2015-07-29:
Addressed by #224.

commit c9a22167 (2015-07-16): Ensures observers can't be added twice. Fixes #110
commit 57bcea3d (2015-07-21): Ensures observers can't be added twice. Fixes #110
commit 7b45a580 (2015-07-23): Ensures observers can't be added twice. Fixes #110
commit 8c3fbefb (2015-07-29): Ensures observers can't be added twice. Fixes #110
closed 2015-07-29 by tcr3dr