dronekit/dronekit-python #111: Number of vehicles connected check

ggregory8, 2015-05-13:
Hi I can't seem to find if there is a way to detect if any vehicles are connected. I can connect with the following commands:
```
api = local_connect()
v = api.get_vehicles()[0]
```
But sometimes I find I lose my serial connection to the UAV. So mavproxy will load but not connect to any vehicle. I then want to detect that in my custom module.

Any current functionality to support this?

hamishwillee, 2015-05-13:
I don't think there is anything explicit in the API. However almost every value within the attributes (e.g. Location, gps_0 etc will have a None value, and if you try to get the mode in this case (ie v.mode) then there will be an exception: ``APIException("Vehicle did not complete initialization")``. So perhaps these could be good indicators?

@mrpollo Ideas? I believe there is a MAVLink heartbeat we should be able to monitor in order to provide an attribute for the connection state?

ggregory8, 2015-05-13:
Thanks for the reply. Yes that's exactly what I resorted to,  checking the mode attribute for None. 

I think adding something to the api is needed. Maybe returning number of vehicles connected for future Multi-vehicle use.

mrpollo, 2015-05-13:
I'm not sure if theres a way to check if the vehicle is still connected but I don't think this would be hard to add ( @hamishwillee can you add an issue for this please? ), as discussed above right now you can check for the ```None``` value on the attribute.

Also a **No Link** check should be a good addition as well

hamishwillee, 2015-05-13:
I've added #112 and #113. These catch the two cases of interest:
* The MAVProxy connection has not been set up properly
* The link on a particular vehicle is working

@ggregory8 You can do ``len(api.get_vehicles())`` to get the count of vehicles and with #112 you will be able to iterate to get the active vehicles. 

@mrpollo a "nice to have" would be to overload get_vehicles such that ``get_vehicles(connected=None)``, where ``connected=None`` means "return all vehicles", ``connected=True`` means "return all connected", ``connected=false`` means "return all inactive" (or similarly have a mechanism so people can just get the active connections. Do you think this is useful? If so I will create an issue for it.

As an aside, we should perhaps add "priority" field to github to let us manage the priority of suggestions a bit better.

hamishwillee, 2015-05-22:
Closing this, as the requirement captured by the new issues

closed 2015-05-22 by hamishwillee