dronekit/dronekit-python #115: No way to disable the callback set_mavlink_callback

hamishwillee, 2015-05-14:
We can add a callback (though it is not recommended) using set_mavlink_callback but there is no way to remove it.

tcr3dr, 2015-07-17:
There's no way simple way to test that this issue causes wrong behavior except by looking at the verbose logs, as indeed it silently fails! But I've added a test that checks for regressions.

The way to "unset" set_mavlink_callback as of #210 is:

```
v.set_mavlink_callback(None)
```

hamishwillee, 2015-07-17:
Thanks @tcr3dr . I will update the documentation when #210 is merged.

hamishwillee, 2015-07-21:
TODO Update docs when #225 is merged (supersedes #210)

tcr3dr, 2015-07-27:
@hamishwillee #240 has been merged which fixes this issue. Documentation was added for `unset_mavlink_callback` but it may need to be improved. Let me know.

hamishwillee, 2015-07-28:
@tcr3dr I've added #245 to improve the documentation. What you had was fine (though you needed a space after the ``..code: python`` line to get the code block to render). The changes add information about the API to the guide and example code. I took the opportunity to improve the related docs.

In addition, I couldn't merge this directly onto the main repo (had to do it via my fork because I got a 403). Hope the approach is OK. 

Yes, this can be closed when minor branch is merged.

tcr3dr, 2015-07-28:
Merged!

hamishwillee, 2015-08-12:
Reopening - this fix will go into Release 1.5.

commit 70637e58 (2015-07-21): Enables set_mavlink_callback to be passed None. Fixes #115
commit 3f2e80bb (2015-07-23): Enables set_mavlink_callback to be passed None. Fixes #115
commit 7a3feba4 (2015-07-27): Enables set_mavlink_callback to be passed None. Fixes #115
commit d3edbb0e (2015-07-27): Merge pull request #240 from dronekit/tcr-115

Enables set_mavlink_callback to be passed None. Fixes #115
closed 2015-08-12 by tcr3dr