dronekit/dronekit-python #116: guided_set_speed_yaw example couldn't work with a real copter vehicle

killerlayup, 2015-05-14:
Hi, all

Recently I ran this 'guided_set_speed_yaw' example in a real copter vehicle. It seems that it can response to 'vehicle.commands.takeoff(5)' command, but it can't work with the 'send_nav_velocity(velocity_x, velocity_y, velocity_z)'. The copter has no response when it receives the command. However, this example can work very well in simulator. This is really confused me.

In 'https://github.com/rmackay9/ardupilot-balloon-finder' project, the Copter can response to command ‘send_nav_velocity(velocity_x, velocity_y, velocity_z)’. 

Is something wrong with my hardware or software settings?  I really need help....

mrpollo, 2015-05-14:
Hi @killerlayup what ArduCopter version where you using on SITL? is that different from the version no your copter?, can you provide us with a log please?

hdmathias, 2015-05-14:
send_nav_velocity depends on set_position_target_local_ned which requires AC3.3. I've used it successfully on an Iris+ running AC 3.3-rc3.

hamishwillee, 2015-05-14:
If you don't want to use AC3.3, it might be possible to set velocity using http://mavlink.org/messages/common#SET_POSITION_TARGET_GLOBAL_INT on earlier versions. I haven't tested this or checked when that went in.

@killerlayup - OK for me to close this?

mrpollo, 2015-05-14:
@hamishwillee I don't think @killerlayup has responded yet, lets see what he's got to say before closing this

hamishwillee, 2015-05-14:
@mrpollo of course - I was just asking if killerlayup considers this answered :-). Note that I could mark the AC version as a requirement in the docs, but I figure that in short order most people will be using this. I will be testing the SET_POSITION_TARGET_GLOBAL_INT next week, and if that works will update the example to use it instead.

killerlayup, 2015-05-17:
Hi, mrpollo
here is my log when I connect my real ArduCopter:
odroid@odroid:~$ mavproxy.py --master=/dev/ttyUSB0 --baudrate 1500000 --aircraft   MyCopter
Connect /dev/ttyUSB0 source_system=255
MyCopter/logs/2013-01-01/flight26
Logging to MyCopter/logs/2013-01-01/flight26/flight.tlog
Running script /home/odroid/.mavinit.scr
-> module load droneapi.module.api
DroneAPI loaded
Loaded module droneapi.module.api
no script MyCopter/mavinit.scr
MAV> fence breach
online system 1
LAND> Mode LAND
APM: ArduCopter V3.3-dev (73e1719e)
APM: PX4: 4cc6120e NuttX: eba6b56f
APM: Frame: QUAD
APM: PX4v2 00300037 30345108 31323737
Flight battery 100 percent
Received 436 parameters
Saved 436 parameters to MyCopter/logs/2013-01-01/flight26/mav.parm

Meanwhile, my companion computer is Odroid XU3, droneapi version is 1.1.1, MAVProxy version is 1,4,15 and pymavlink version is 1.1.51.  what the problem it will be??

In SITL, everything goes well, it really confused me.

killerlayup, 2015-05-17:
Hi, @mrpollo @zlite @eliao @hamishwillee @hdmathias

The copter can response the 'takeoff' command. I check the 'takeoff' command and find it is defined in api.py:

    def takeoff(self, alt=None):
        if alt is not None:
            altitude = float(alt)
            self.__module.master.mav.command_long_send(self.__module.target_system,
                                                    self.__module.target_component,
                                                    mavutil.mavlink.MAV_CMD_NAV_TAKEOFF,
                                                    0, 0, 0, 0, 0, 0, 0,
                                                    altitude)

it seems that the copter can response to  command_long_send().

hamishwillee, 2015-05-17:
@killerlayup FYI, the "Command long send" is just a package for one of the other commands - so the actual command that is processed here is MAV_CMD_NAV_TAKEOFF (and yes [it is supported](http://planner.ardupilot.com/wiki/common-planning-a-mission-with-waypoints-and-events/common-mavlink-mission-command-messages-mav_cmd/#copter-2) )

What mode were you in when you tried to take off? If you were in LAND as shown I would not be surprised if the commands were ignored. What happens if you take off and then run the script?

killerlayup, 2015-05-18:
Hi, @mrpollo @zlite @eliao @hamishwillee @hdmathias

I just ran the 'guided_set_speed_yaw' example.  The script sets mode 'STABILIZE' and armed, then  
sets mode 'GUIDED' and takeoff, then run the command 'condition_yaw' and 'send_nav_velocity'.

when the script runs, the copter can takeoff to the desired altitude but then the copter only hover and didn't response to the following command. But the terminal shows that the script has ran the  command 'condition_yaw' and 'send_nav_velocity'.

Thanks.

hdmathias, 2015-05-18:
What is the value of the mask (or is it flag?) parameter to your set_position_target_local_ned command in send_nav_velocity?  If it's 0, that is likely why that command isn't working. Unfortunately, it doesn't explain why condition_yaw fails to work.

Interesting that you can takeoff in GUIDED mode. Since uploading AC3.3, I can't. What firmware are you using?

killerlayup, 2015-05-18:
Hi, @hdmathias  

the flag is 0b0000111111000111 just like the example. 
my copter version:
APM: ArduCopter V3.3-dev (73e1719e)
APM: PX4: 4cc6120e NuttX: eba6b56f
APM: Frame: QUAD
APM: PX4v2 00300037 30345108 31323737
Flight battery 100 percent

pls help me .

thanks

killerlayup, 2015-05-18:
Hi, @hdmathias 

what's your firmware version?

mrpollo, 2015-05-18:
@killerlayup at this point I would need to see your code, can you post a gist please?

killerlayup, 2015-05-18:
@mrpollo  

which part of code do you need to see?

mrpollo, 2015-05-18:
the offending code that caused you to open this issue please

hdmathias, 2015-05-18:
I'm running 3.3-rc3. Your mask is the same as what I use.

killerlayup, 2015-05-18:
@mrpollo  
I just ran the  'guided_set_speed_yaw' example in this repository.  
I think maybe my firmware causes the issue. 
I download my firmware in site http://firmware.diydrones.com/Copter/beta/PX4-quad/
Is this right way to download the firmware?

killerlayup, 2015-05-18:
Hi @hdmathias 

would you pls provide the 3.3-rc3 firmware download site address?

thanks!

hdmathias, 2015-05-18:
I install firmware using MissionPlanner. On the firmware tab there's a link for beta versions.

hdmathias, 2015-05-18:
I just looked at that example. I think I read somewhere that there's a problem with z velocities in the set_position_target_local_ned command. I've always used 0 for the z velocity so I can't confirm or refute that.

mrpollo, 2015-05-18:
@killerlayup we don't yet fully support PX4, we only test using Ardupilot https://github.com/diydrones/ardupilot

hdmathias, 2015-05-18:
@mrpollo so it works with z velocities on a real vehicle?  Good to know -- I've been afraid to test it.

killerlayup, 2015-05-18:
@mrpollo 

well, where should download and install the fireware on a real vehicle?
using Mission Planner or some other way?

killerlayup, 2015-05-20:
Hi, guys

I update the AC firmware and get everything right!

thank you guys!

closed 2015-05-20 by killerlayup

hamishwillee, 2015-05-20:
Awesome! Thanks for letting us know