dronekit/dronekit-python #121: Failed to load module: No module named google.protobuf.internal

ggregory8, 2015-05-15:
Hi I'm getting a ```Failed to load module: No module named google.protobuf.internal``` error when trying to load the droneapi module in mavproxy.

I did have the droneapi working but because I installed as root it put the .mavinit.scr file in /home/root. I since set up a user account. I then uninstalled the droneapi package then reinstalled from px4 login with sudo. Now the droneapi isn't working and giving the above mentioned error.

Obviously the uninstall / install has messed it up somehow but can't see where. The output at the bottom shows there is multiple instances of the api.py file. Maybe the path to the correct location is missing?

Any ideas how to fix this?

Below is the traceback when trying to load droneapi in mavproxy.
````
GUIDED> Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 260, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 390, in import_package
    mod = __import__(name)
ImportError: No module named mavproxy_droneapi.module.api

Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 260, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 390, in import_package
    mod = __import__(name)
  File "/usr/local/lib/python2.7/dist-packages/droneapi/module/api.py", line 8,                in <module>
    from droneapi.lib.WebClient import *
  File "/usr/local/lib/python2.7/dist-packages/droneapi/lib/WebClient.py", line                1, in <module>
    from webapi_pb2 import *
  File "/usr/local/lib/python2.7/dist-packages/droneapi/lib/webapi_pb2.py", line                4, in <module>
    from google.protobuf.internal import enum_type_wrapper
ImportError: No module named google.protobuf.internal

Failed to load module: No module named google.protobuf.internal. Use 'set moddebug 3' in the MAVProxy console to enable traceback
```

Below is output to show where the api.py file is located, in many locations:
```
px4@ubilinux:~$ sudo find / -iname *api.py*
/usr/share/pyshared/numpy/core/generate_numpy_api.py
/usr/share/pyshared/numpy/core/tests/test_api.py
/usr/lib/pymodules/python2.6/numpy/core/generate_numpy_api.py
/usr/lib/pymodules/python2.6/numpy/core/generate_numpy_api.pyc
/usr/lib/pymodules/python2.6/numpy/core/tests/test_api.py
/usr/lib/pymodules/python2.6/numpy/core/tests/test_api.pyc
/usr/lib/pymodules/python2.7/numpy/core/generate_numpy_api.py
/usr/lib/pymodules/python2.7/numpy/core/generate_numpy_api.pyc
/usr/lib/pymodules/python2.7/numpy/core/tests/test_api.py
/usr/lib/pymodules/python2.7/numpy/core/tests/test_api.pyc
/usr/local/lib/python2.7/dist-packages/requests-2.7.0-py2.7.egg/requests/api.pyc
/usr/local/lib/python2.7/dist-packages/requests-2.7.0-py2.7.egg/requests/api.py
/usr/local/lib/python2.7/dist-packages/pymavlink/generator/lib/genxmlif/xmlifApi.py
/usr/local/lib/python2.7/dist-packages/pymavlink/generator/lib/genxmlif/xmlifApi.pyc
/usr/local/lib/python2.7/dist-packages/droneapi/module/api.pyc
/usr/local/lib/python2.7/dist-packages/droneapi/module/api.py
/root/droneapi-python/droneapi/module/api.py
/root/build/pymavlink/build/lib.linux-i686-2.7/pymavlink/generator/lib/genxmlif/xmlifApi.py
/root/build/pymavlink/generator/lib/genxmlif/xmlifApi.py
/root/build/droneapi/build/lib.linux-i686-2.7/droneapi/module/api.py
/root/build/droneapi/droneapi/module/api.py
/home/px4/build/droneapi/droneapi/module/api.py
/home/px4/build/droneapi/build/lib.linux-i686-2.7/droneapi/module/api.py
```

codywburns, 2015-05-16:
I am also having the same problem. I attempted to manually install python-protobuf, but I still get an error that it is missing enum_type_wrapper when loading mavproxy.

codywburns, 2015-05-16:
I found the fix.
Mine was failing during the pip install phase because it was not finding "../src/google/protobuf/unittest.proto" 
The protobuff git put out a fix 9 days a go that involved adding an if statement in the setup.py file to ignore the unittest section if you are not working on dev. Worked like a charm!

https://github.com/google/protobuf/commit/e4f4d9fe83887b08b0db67512947645324826755

hamishwillee, 2015-05-18:
@mrpollo - assigned to you - something to fix in our installation?

mrpollo, 2015-05-18:
@codywburns thanks for the report, can you create a different issue for your problem please? even if its solved already

codywburns, 2015-05-18:
no problem

mrpollo, 2015-05-18:
@ggregory8 did you uninstall both ```droneapi``` and ```mavproxy```?, also when you installed them again under your user **px4** did you re/install both using ```sudo```?, do you know the contents of ```$PYTHONPATH```?

```
echo $PYTHONPATH
```

ggregory8, 2015-05-20:
@mrpollo Sorry about delay, have been away. Yes I had uninstalled both droneapi and mavproxy and reinstalled using sudo. Just tried again with same results.

I noticed during pip install that I am getting same error as @codywburns. So I followed his advice as per issue #125 and it is now working.

Also ```echo $PYTHONPATH``` hasn't showed anything, even before I was having this issue. Is there another way to check this?

hamishwillee, 2015-07-23:
This should be fixed on windows at least by #228

tcr3dr, 2015-10-14:
This will no longer be an issue since https://github.com/dronekit/dronekit-python/pull/341.

closed 2015-10-14 by tcr3dr