dronekit/dronekit-python #122: Ensure that Companion Computer docs or DK docs cover installation of files to UAV.

hamishwillee, 2015-05-16:
Make sure the docs explain how to install and run scripts on a real device.

mrpollo, 2015-05-18:
I think putting your code into a companion board should be a given and everyone should handle it as their setup best allows them to on a DIY setup, if we are talking specifically about Solo then it should be noted that Solo has many components and that it probably is going to be a different setup to what is currently out there, I think we would need to offload this to the Solo SDK docs on how to move dronekit specific files and how to run them on Solo.

TL;DR; Agreed, lets wait until Solo is shipped and the official SDK is out

hamishwillee, 2015-05-18:
Personally I think mid-term we should integrate with Mission Planner or Tower to handle the upload - possibly with the DroneKit setup configuring the companion computer to support the upload method. 

More than happy to wait on Solo.