dronekit/dronekit-python #123: Add section/link in Debugging topic to observers docs about logging

hamishwillee, 2015-05-18:
When Debugging topic is live and attribute observers "fixed" it makes sense to link to the attribute observer topic.

Perhaps could put in example of a "generic" observer.
```
# Another callback example - this time "generic"
def attribute_callback(attribute):
    print "Generic callback - '%s' changed: %s" % (attribute, getattr(v, attribute) )
	

# Set callback observer for mode
v.add_attribute_observer('armed', attribute_callback)	
```
We should also put in link to the mavlink message callback, but only when API exists to disable it.

mrpollo, 2015-05-18:
What would you say is a good use for a generic observer?

hamishwillee, 2015-05-18:
It would be useful if you want to handle all callbacks in exactly the same way. Off the top of my head, the only thing I can think of is logging, hence this post. So code below is much more compact than separate duplicate callbacks.

```
# Another callback example - this time "generic"
def attribute_callback(attribute):
    print "Generic callback - '%s' changed: %s" % (attribute, getattr(v, attribute) )

# Enable the ones you want
v.add_attribute_observer('armed', attribute_callback)  
# v.add_attribute_observer('location', attribute_callback)  
# v.add_attribute_observer('mode', attribute_callback)  
v.add_attribute_observer('velocity', attribute_callback)  
....
``