dronekit/dronekit-python #132: Vehicle.commands is throwing away the first command sent

hamishwillee, 2015-05-20:
The first command set in the mission is being discarded (probably by DroneKit). The following code shows how to create  a number of missions, starting with takeoff.

```
#add a takeoff command
altitude=100 #target altitude
pitch=45 #take off pitch. Need to check if degrees or radians, and what is a reasonable valued.
cmd = Command(  0, 0, 0, mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, 
                mavutil.mavlink.MAV_CMD_NAV_TAKEOFF, 0, 0, 
                pitch, 0, 0, 0, 0, 0, altitude)
cmds.add(cmd)
			   
lat = -35.363768
lon = 149.154275
altitude = 110
cmd = Command(0,0,0, mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, mavutil.mavlink.MAV_CMD_NAV_WAYPOINT,
    0, 0, 0, 0, 0, 0,
    lat, lon, altitude)
cmds.add(cmd)

lat = -35.3666207
lon = 149.1662800
altitude = 120
cmd = Command(0,0,0, mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, mavutil.mavlink.MAV_CMD_NAV_WAYPOINT,
    0, 0, 0, 0, 0, 0,
    lat, lon, altitude)
cmds.add(cmd)

lat = -35.363769
lon = 149.154276
altitude = 130
cmd = Command(0,0,0, mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, mavutil.mavlink.MAV_CMD_NAV_LOITER_UNLIM,
    0, 0, 0, 0, 0, 0,
    lat, lon, altitude)

cmds.add(cmd)
v.flush()
```

The output below shows the missions as sent, and the missions that are then re-retrieved from the vehicle. The first mission item is simply discarded (the first mission item is now the home)

```
AUTO> Create mission
Got MAVLink msg: MISSION_ACK {target_system : 255, target_component : 0, type : 1}
Sent waypoint 0 : MISSION_ITEM {target_system : 1, target_component : 0, seq : 0, frame : 3, command : 22, current : 0, autocontinue : 0, param1 : 45, param2 : 0, param3 : 0, param4 : 0, x : 0, y : 0, z : 100}
Sent waypoint 1 : MISSION_ITEM {target_system : 1, target_component : 0, seq : 1, frame : 3, command : 16, current : 0, autocontinue : 0, param1 : 0, param2 : 0, param3 : 0, param4 : 0, x : -35.363768, y : 149.154275, z : 110}
Sent waypoint 2 : MISSION_ITEM {target_system : 1, target_component : 0, seq : 2, frame : 3, command : 16, current : 0, autocontinue : 0, param1 : 0, param2 : 0, param3 : 0, param4 : 0, x : -35.3666207, y : 149.16628, z : 120}
Sent waypoint 3 : MISSION_ITEM {target_system : 1, target_component : 0, seq : 3, frame : 3, command : 17, current : 0, autocontinue : 0, param1 : 0, param2 : 0, param3 : 0, param4 : 0, x : -35.363769, y : 149.154276, z : 130}
Sent all 4 waypoints
Got MAVLink msg: MISSION_ACK {target_system : 255, target_component : 0, type : 0}
APM: flight plan received

CHECK: Getting commands
Requesting 4 waypoints t=Wed May 20 17:58:44 2015 now=Wed May 20 17:58:44 2015
commands
Command MISSION_ITEM {target_system : 255, target_component : 0, seq : 0, frame : 0, command : 16, current : 0, autocontinue : 1, param1 : 0.0, param2 : 0.0, param3 : 0.0, param4 : 0.0, x : -35.3632621765, y : 149.165237427, z : 584.099975586}
Command MISSION_ITEM {target_system : 255, target_component : 0, seq : 1, frame : 3, command : 16, current : 0, autocontinue : 1, param1 : 0.0, param2 : 0.0, param3 : 0.0, param4 : 0.0, x : -35.3637695312, y : 149.154281616, z : 110.0}
Command MISSION_ITEM {target_system : 255, target_component : 0, seq : 2, frame : 3, command : 16, current : 0, autocontinue : 1, param1 : 0.0, param2 : 0.0, param3 : 0.0, param4 : 0.0, x : -35.3666191101, y : 149.166275024, z : 120.0}
Command MISSION_ITEM {target_system : 255, target_component : 0, seq : 3, frame : 3, command : 17, current : 0, autocontinue : 1, param1 : 0.0, param2 : 0.0, param3 : 1.0, param4 : 0.0, x : -35.3637695312, y : 149.154281616, z : 130.0}
cmds count: 4
cmds next: 4
```

This is a serious bug! I suspect that someone thought that since copter ignores the takeoff command in AUTO it could be removed, but testing shows that it removes any command irrespective of type. Also Plane doesn't ignore the command. 

In the short term someone could work around this by adding the takeoff command twice.

mrpollo, 2015-05-20:
Thanks for the report @hamishwillee I'll look into this.

I think this also brings up another topic we haven't properly discussed "Unit Tests", if we have a setup for unit tests we can probably tests this behavior and apply a fix/test, but lets create another issue for that

hamishwillee, 2015-05-21:
Hi @mrpollo 

Some simple test code showing the problem (visual inspection only) can be found  [here](https://github.com/hamishwillee/dronekit-python/blob/withhamishtests/hamishtests/test_commands.py).

Yes, we do need to talk about Unit tests, and that would have found this problem. I don't think some of this will be easy though because even simple testing like checking arming can be dependent on Mode and other factors. It would be really nice to have a compatibility suite too.

Hope you can fix this soon, as examples with workarounds are ugly!

Cheers
H

tcr3dr, 2015-07-16:
Hey @hamishwillee, I adapted your test case to make a unit test! https://github.com/diydrones/dronekit-python/commit/1a114948ac0fc1099cbdd13cb6b1b3cbae558997

So if I understand the issue (which still exists) properly:

* Clearing a waypoint list should result in one waypoint, "home"
* Adding waypoints, then uploading, then downloading, should result in an identical list

And what appears to be happening:

* Clearing a waylist list without flushing collides with...
* Adding way points, then uploading, which...
* Results in a list with one item too few: the first "added" item is overwritten by the "home" waypoint

Specifically, the `v.flush()` line I added to this test makes it behave properly: https://github.com/diydrones/dronekit-python/blob/1a114948ac0fc1099cbdd13cb6b1b3cbae558997/tests/sitl/test_132.py#L32 If the line is removed, we observe the behavior as seen in your test case.

Does this sound right? With a concrete test I can delve into why this is happening.

hamishwillee, 2015-07-16:
@tcr3dr You are correct in every detail - thanks very much. Clearly what I was doing was a mis-use of the API - I will improve the documentation. 

That said, with other commands you can normally batch them up and then call flush to make sure they are sent - so sending a mode change and arm request before calling flush works. Others will make this mistake and it would be good if our API was robust enough to help avoid this problem. Not a high priority though, so I will change label above. 

As an aside/FYI the use of invisible waypoint "0" as the home point is apparently a hack that will be undone at some point.

hamishwillee, 2015-07-28:
@tcr3dr @mrpollo I've run into another issue from this behaviour, which you can find out about in https://github.com/dronekit/dronekit-python/issues/230#issuecomment-125521588.

IMO the change that needs to happen is that DK-Python makes the 1st mission item (Home location) invisible to user through this API - and only ever hands it a CommandSequence from the first "real" waypoint. Under the hood the API should make sure that any commands are always added above the home location.

We would of course need another way to get/set the home location - I suggest an attribute.

tcr3dr, 2015-07-28:
@hamishwillee Let's fix this properly for the minor branch then. `Vehicle.commands` pairs with `Vehicle.home` or similar. I'll start developing this on my existing PR branch and open a new issue about it.

hamishwillee, 2015-07-28:
Sounds great.

jamesbellaero, 2015-08-07:
Something that I've noticed is that when I clear, flush, add new waypoints, and flush again is that the Home location is set to the first standard waypoint (not takeoff or change speed, etc) in the command sequence. I've created a workaround by just adding the Home waypoint after I clear the list, but I'm gathering from reading this and the Dronekit.io documentation that this is a bug. 

Just letting y'all know about it. I can replicate the problem if you need it. My last pull request was a few days ago, so this may have been solved since then anyway.

hamishwillee, 2015-08-09:
Hi @jamesbellaero 

Yes, if you clear the vehicle.commands then you clear out all waypoints including the home position. When you upload, the bug reported here means that the first command you added will be lost. There are two solution - either add a dummy waypoint first or do a download after clearing in order to populate the first waypoint. I submitted https://github.com/dronekit/dronekit-python/pull/260/ (not published yet) which is supposed to add some clarity on this subject. 

The implementation of the autopilot to use the first waypoint as the home position is a misinterpretation of the protocol. In any case though, a user should not need to think about this when using the API. Our plan to address this is to make the API hide the command - a user will have an api to access and handle home, and a separate API to handle missions. https://github.com/dronekit/dronekit-python/issues/252

Thanks for reporting this.

hamishwillee, 2015-11-02:
Fixed by #384

closed 2015-11-03 by tcr3dr