dronekit/dronekit-python #133: Methods to get distance to target and from home

hamishwillee, 2015-05-22:
It is useful to be able to perform actions based on distance to target or distance from home. It would also be useful to be able to get a callback when the distance from target/home is a certain value.

While the maths for determining distances is fairly straighforward, this is probably required often enough that providing an API makes sense.

hamishwillee, 2015-06-01:
Acutally, more generically would probably be good to be able to shove in any two Location objects and get the distance (maybe both as single number at ground level (ie x, y plane) and as an x, y, z)

jmacgithu, 2015-06-04:
More generally, there are a bunch of functions that could be imported from the ardupilot autotest [common.py] (https://github.com/diydrones/ardupilot/blob/master/Tools/autotest/common.py) file as they might be useful... I.e. `get_distance(loc1, loc2)` and the `wait_*` ones...

hamishwillee, 2015-06-05:
@jmacgithu yes, they look very useful. @mrpollo thoughts - best way to integrate?

hamishwillee, 2015-06-24:
#170 adds some documentation for the more useful of these, so they will be in the guide shortly. I propose we close this as sorted when those changes go live.

hamishwillee, 2015-06-26:
@mrpollo What do you think (see previous comment). Should we close this, or should we keep it open and plan integration of helper functions into the API?

mrpollo, 2015-06-26:
we should implement

hamishwillee, 2015-07-07:
@mrpollo I agree.

My only concern is that most of the algorithms floating around (like those I documented) are only good within certain latitudes, and over certain distances. With the "documentation solution" you can just add a warning and people can try alternatives if they find errors. 

If you provide a function then you need to make it far more reliable over larger distances/latitudes or provide some sort of "accuracy" feedback. This is particularly true if we start to rely on it - for example I can see we might want to provide our own "frames of reference code" and that would have to depend on the library for conversions.

Another thought - is there a good 3rd party library we could either use or recommend instead of rolling out our own solution.