dronekit/dronekit-python #136: Build seems to have reverted to old reference and theme

hamishwillee, 2015-05-28:
The build isn't right after #124 - unrelated to that particular push though.
See http://python.dronekit.io/automodule.html - this is the old API reference (like months!)

Also there are big spaces above the level 2 headings, which were removed by current theme.

Something not right!

tcr3dr, 2015-07-16:
@hamishwillee Can you confirm this is an issue? I want to know if this is something I can dig into.

hamishwillee, 2015-07-16:
@tcr3dr Not an issue right now so I will close. It reverted a few times in the past but @mrpollo did some changes which may have fixed it. If this occurs again I'll open a generic issue about the fact it keeps reverting.

closed 2015-07-16 by hamishwillee

tcr3dr, 2015-07-16:
@hamishwillee Sounds good. I'll be able to help "in the moment" fix it if this comes up.