dronekit/dronekit-python #137: Check documentation for new battery feature

hamishwillee, 2015-05-29:
The new battery featured added in #135 has API reference. Action is to check the reference and also make sure that this feature is documented in the guide and examples where relevant - e.g. http://python.dronekit.io/guide/vehicle_state_and_parameters.html

maruina, 2015-05-29:
Do you have guidelines for writing the documentation? I can fix the documentation for the battery object.

hamishwillee, 2015-05-29:
Hi @maruina 
Probably easier for me to do this. It isn't because I don't think you can't do it, its because I just completed the sections I want to update so I know I can do it very quickly. Thank you

maruina, 2015-05-29:
Sure, no problem!

hamishwillee, 2015-06-01:
Hi @maruina @mrpollo 

I've created [draft documentation](https://github.com/hamishwillee/dronekit-python/tree/tmp-battery-docs) which you [can view here](https://dl.dropboxusercontent.com/u/3067678/share2/dronekit-python/html/automodule.html#droneapi.lib.Vehicle.mount_status).

Some questions:

1. I refer to this as "system" battery information, based on the fact it comes from [SYSTEM_STATUS](https://pixhawk.ethz.ch/mavlink/#SYS_STATUS). Is this in fact the combined system battery information for all possible batteries, or is this just for the primary battery?

2. For battery level the text is now "battery_level – Remaining battery level as a percentage (0-100). -1 if the Autopilot should estimate the remaining battery energy." 

What does this mean? I think that the spec is "confused" and what this means that the autopilot does NOT measure/estimate the remaining charge and the caller of the message will have to. Thoughts?

@mrpollo 
1. This will require further changes if you agree we should change the API to return ```None``` for error/not supported (see #135) and based on answers to above.
2. This can't go live until you'd updated PIP  - otherwise the examples will fail because the updated version refer to battery, which is not available in the published API.
3. What version of droneAPI will the next version be? 1.1.2 or 1.2.0? Should I add "introduced in" tagging for this new method (?)

Regards
H

hamishwillee, 2015-06-01:
@mrpollo @maruina 
I just tested the battery call by modifying the vehicle_state.py example to add the line:
```
print " Battery: %s" % v.battery
```

This failed in  new code as shown below (thoughts please):

```
Exception in Thread-1: float() argument must be a string or a number
Traceback (most recent call last):
  File "build/bdist.linux-i686/egg/droneapi/module/api.py", line 325, in run
    self.fn()
  File "build/bdist.linux-i686/egg/droneapi/module/api.py", line 605, in <lambda>
    APIThread(self, lambda: execfile(args[1], g), args[1])
  File "../../dronekit-python/example/vehicle_state/vehicle_state.py", line 27, in <module>
    print " Battery: %s" % v.battery
  File "build/bdist.linux-i686/egg/droneapi/module/api.py", line 179, in battery
    return Battery(self.__module.voltage, self.__module.current, self.__module.level)
  File "build/bdist.linux-i686/egg/droneapi/lib/__init__.py", line 181, in __init__
    self.voltage = float(voltage) / 1000.0
TypeError: float() argument must be a string or a number
```

commit 3a9f399c (2015-06-02): Merge pull request #139 from maruina/master

Fix #137
commit 6951a3a2 (2015-06-02): Fix #137
closed 2015-06-02 by mrpollo