dronekit/dronekit-python #151: DroneKit's goto() function should have a local/relative option vice only using absolute coordinates

a7scenario, 2015-06-04:
I want to call goto() and have it move the drone "20 meters north of current position".  If I'm not mistaken, to do this in DroneKit, I think you have to either first figure out a way to convert your local command ("20 meters north") to absolute lat/lon OR use DroneKit's messagefactory to build a regular mavlink message.  I think relative movements of the drone is something that would warrant a simplified, higher-level helper function in DroneKit.  

Also, the "is_relative" input variable to DroneKit's goto() function is a little deceiving that way it is named.  It only applies to altitude, not lat/lon.

mrpollo, 2015-06-04:
agreed that ```is_relative``` is somewhat deceiving we need to make sure this works as expected.

```SET_POSITION_TARGET_LOCAL_NED``` can be used to move 20 meters if you set it to ignore accel and velocity by using a type_mask for that

mrpollo, 2015-06-04:
@rmackay9 can confirm

hamishwillee, 2015-06-05:
The is_relative is documented correctly. I'm not sure how you would "fix" it because most NAV commands expect the frame to be MAV_FRAME_GLOBAL_RELATIVE_ALT (ie only the ALT is relative). I guess you could make all values relative, but you'd need to translate what is coming back from the autopilot.

>SET_POSITION_TARGET_LOCAL_NED can be used to move 20 meters if you set it to ignore accel and velocity by using a type_mask for that

My understanding is that this is correct. There is an example of the VELOCITY being set in the [examples](http://python.dronekit.io/examples/guided-set-speed-yaw-demo.html) to show you the "idea". I will soon (maybe!) be adding a "guided mode" section to the docs that demonstrates this.

EmirhanKarahan, 2021-08-16:
year 2021, is there any way to do it? i mean i just want to move the drone 20 meters north, south, east, west but relative to the heading

EmirhanKarahan, 2021-08-16:
you can use SET_POSITION_TARGET_LOCAL_NED with MAV_FRAME_BODY_OFFSET_NED option.  see [goto_position_target_local_ned()](https://dronekit-python.readthedocs.io/en/latest/examples/guided-set-speed-yaw-demo.html) and [other frame options](https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html)