dronekit/dronekit-python #152: Installing dronekit-python(windows) instructions require admin access

djnugent, 2015-06-04:
The docs for droniekit-python windows should specify that the user run the winpython installer as an admin and that they run command-prompt as an admin. If the user is not admin they will get stuck very quickly.

Also a note about the winpython install location: If you open the installer from your downloads folder then it defaults to install in the downloads folder unless the user specifies another location. We should specify that the user install in the programs(x86) folder and then tell them to navigate there to access the winpython control panel.

djnugent, 2015-06-04:
I was also unable to run mavproxy properly with winpython 3.4.3.3. I used 2.7 instead http://sourceforge.net/projects/winpython/files/WinPython_2.7/2.7.9.5/

mrpollo, 2015-06-04:
@djnugent i think WinPython version numbers are matching Python version numbers, we currently don't support 3 yet, 2.7 should be fine.

djnugent, 2015-06-04:
@mrpollo I made a mistake. I just clicked on the link again and I see where I went wrong. Winpython recommends to download the most recent version, so I clicked that before I even saw the 2.7 download lower on the page. Oops

hamishwillee, 2015-06-05:
I have updated the quickstart to explicitly recommend Python v2.7. This should be fixed by https://github.com/diydrones/dronekit-python/pull/149

djnugent, 2015-06-09:
Another issue with the quickstart: Windows does not come with a ssh client installed, this makes "vagrant ssh" throw an error.
Solution:
Download Putty
Connect to 127.0.0.1:2222
username: vagrant
pass: vagrant

djnugent, 2015-06-09:
Step 7 in the quick start instructs the user to configure mavproxy using "echo... >> some file"
However, prior to step 7, there is no mention of mavproxy and how to install it.

mrpollo, 2015-06-09:
@djnugent can you break this into separate issues, it's ok to keep discussing windows only instructions here

djnugent, 2015-06-09:
Oh yeah forgot this was windows and not quick start oops.

hamishwillee, 2015-06-11:
I think this is addressed, or will be by #155. OK to close?

djnugent, 2015-06-11:
Do we want to add anything about admin access or install location. Can anyone verify my claims?

hamishwillee, 2015-06-11:
Adding the admin and install stuff to #155 and closing this.

closed 2015-06-11 by hamishwillee