dronekit/dronekit-python #153: Fix layout of loading API

hamishwillee, 2015-06-09:
In [Getting Started](http://python.dronekit.io/guide/getting_started.html)

[image]

hamishwillee, 2015-06-09:
@mrpollo The source file here is correct, and so is the HTML. Unfortunately the paragraph is not being rendered:
[image]

So this is a theme/CSS problem. As I still can't build the theme, handballing to you!

zlite, 2015-06-11:
Any update on this? Seems nuts that we have to go through backflips to add a carriage return

[image]

mrpollo, 2015-06-11:
our great @kaitlynhova is taking a look at this, she has some fixes in place right now and will push once she feels she's got them all sorted, I'll post back with update.

hamishwillee, 2015-06-11:
@zlite @mrpollo This is now fixed on the live version. Thank you @kaitlynhova

closed 2015-06-11 by hamishwillee