dronekit/dronekit-python #154: is dronekit-python api only for Mavproxy?

sky129225, 2015-06-09:
I wonder if there are any other use cases for droneapi. and if I don't want to use mavproxy, does it mean I have to write my own mavlink layer? and how can i do it

mrpollo, 2015-06-09:
Hi @sky129225 there have been talks of moving the core of MAVProxy into a library and use that instead as to have DroneKit be a standalone, work on that hasn't been made but we would be happy to guide you through the changes needed if you feel like working on it makes sense to you, come find us on Gitter https://gitter.im/diydrones/dronekit-python and or reply here

sky129225, 2015-06-10:
I wonder if there are any other use cases for droneapi. and if I don't want to use mavproxy, does it mean I have to write my own mavlink layer? and How do I start to Compile the code?
 api = local_connect()
vehicle = api.get_vehicles()[0]
 i think i just to compile the above code to create the new mavlink layer
 my Program is use "mavutil.mavlink_connection(port, baud, autoreconnect=True)"As connection with drone how can i add in ?

hamishwillee, 2015-06-10:
Hi Sky

> I wonder if there are any other use cases for droneapi. 

What use cases do you have in mind?

> and if I don't want to use mavproxy, does it mean I have to write my own mavlink layer? 

Yes it does. 

As @mrpollo says, you will need to discuss the detail with us, and probably gitter is the place to do that.

> How do I start to Compile the code?

The detailed instructions are still waiting a pull request #149 to be merged. In summary you need to clone this repo, and then you can build using the following commands (on linux)
```
sudo python setup.py build
sudo python setup.py install
```

> api = local_connect()
vehicle = api.get_vehicles()[0]
i think i just to compile the above code to create the new mavlink layer
my Program is use "mavutil.mavlink_connection(port, baud, autoreconnect=True)"As connection with drone how can i add in ?

I don't really understand what you mean. The above instruction if run in a script inside mavproxy will fetch you a vehicle. BUT, you for this to work you need to be running inside MAVProxy, as this is the tool that actually sends the message.

Or to put it in another way, 

1. MAVProxy is the MAVLink implementation used by DroneKit
2. DroneKit is written as a MAVProxy module.

So to use DroneKit without MAVProxy you would have to write a new layer for doing all the MAVLink communication, and write DroneKit to use it. I hope that makes sense.

hamishwillee, 2015-06-10:
@sky129225 The topics on how to build the source [can now be found in the documentation here](http://python.dronekit.io/contributing/index.html)

hamishwillee, 2015-07-08:
@tcr3dr I think this has been answered as far as it can be and should be closed. 

I believe there is a long term strategic objective to to allow DK to be used as "just another Python Module" (possibly by modifying MAVProxy so that it can be loaded as a module). Do you think it reasonable to create a separate issue for this? (and if so, can you do so). I think it would be useful for the community to be able to track our efforts in this area.

tcr3dr, 2015-07-09:
@hamishwillee Sounds good.

closed 2015-07-09 by tcr3dr