dronekit/dronekit-python #155: Quick Start bugs

djnugent, 2015-06-09:
1. Make it more clear that quick start is explicitly windows only. Don't even mention other operating systems. Or have a quick start for as OS's. 

2. Step 5 does not work with native windows because windows does not have a SSH client installed
Solution: Install putty, Connect to 127.0.0.1:2222, username: vagrant, password: vagrant

3. Step 7 in the quick start instructs the user to configure mavproxy using "echo... >> some file"
However, prior to step 7, there is no mention of mavproxy and how to install it.

Personal note: Quick start seems like a non informative version of the guide. I would prefer to just have guides

tcr3dr, 2015-06-09:
I would second the proposal of creating a unified Getting Started / Quick Start guide, or ideally, a Quick Start section composed of specific instructions for each OS (Windows, Linux, OS X) or setup (Companion Computer) Dronekit is targeting. This moves technical rigor and information to later sections of the guide, as well as generally improving what we've seen of user testing so far.

hamishwillee, 2015-06-11:
@tcr3dr @djnugent A quick start is a set of focused instructions to set up DK for running scripts in the the shortest time/effort. It avoids most of the explanation of how and why you do things because the aim is not to show people how to configure for different systems, but to have an environment they can play with very quickly to test script ideas and to test our examples.

In the original version this really was a quick start - DK installed on top of the vagrant instance. This meant you could run scripts in about half the instructions. Of course when you want to move to using companion computers you need to know more, but most users who want to get started don't have a companion computer and just want to see how things work.

Unfortunately there was opposition to this idea (or maybe I didn't sell this right) so now our quick start is a bit of a half-arsed DroneKit on Windows getting started. 

I can certainly remove it/merge it. I am not yet sure if I will do the split yet as suggested - I don't like the idea of duplication. 

@djnugent I will look at these below now:

> Make it more clear that quick start is explicitly windows only. Don't even mention other operating systems. Or have a quick start for as OS's.

Good catch. I'll think about merging, if not I'll rename to Windows Quickstart and make this obvious.

> Step 5 does not work with native windows because windows does not have a SSH client installed
Solution: Install putty, Connect to 127.0.0.1:2222, username: vagrant, password: vagrant

Solution is actually to add SSH exe to your path. I'll add that.

> Step 7 in the quick start instructs the user to configure mavproxy using "echo... >> some file"
However, prior to step 7, there is no mention of mavproxy and how to install it.

I'll think about this. The instructions are like this because you don't nee to know what this is doing in a quick start. Also because you don't need to know about MAVProxy to install it as it comes with Dronekit automatically. 

You make a good point though, I think we need an explanation of the "bits" of the solution before we explain the steps.

Thanks. I'll post again when I push a fix.

hamishwillee, 2015-06-11:
@djnugent @mrpollo @tcr3dr Please check out [this proposed update on my drop box](https://dl.dropboxusercontent.com/u/3067678/share2/dronekit-python/html/guide/getting_started.html) to address the above concerns (it is not 100% done, I'm just after feedback). 

The changes are that I have 

* removed the QuickStart
* Created a new topic for [setting up SITL](file:///C:/Users/hamis_000/Documents/vagranttesting/dronekit-python/docs/_build/html/guide/sitl_setup.html) from the QuickStart. This RECOMMENDS the native setup of SITL for each platform. It includes the section on setting up an additional connection to SITL from a GCS that was formerly in the Getting Started
* [Getting Started](https://dl.dropboxusercontent.com/u/3067678/share2/dronekit-python/html/guide/getting_started.html) is much cleaner - simpler links in the section on setting up vehicles, and removed the end section which was really only relevant to SITL users.

This is much cleaner IMO. Thoughts?

djnugent, 2015-06-11:
@hamishwillee looks great! Much more clear.

hamishwillee, 2015-06-11:
@djnugent Thanks very much. We may have to hack it more as this evolves, but I think separating out the vehicle simulation stuff does make this more clear.

I will tidy, address https://github.com/diydrones/dronekit-python/issues/152#issuecomment-111231852 (admin and location stiff) and then submit a push request.

hamishwillee, 2015-06-11:
@djnugent OK, addressed the run as admin/where to install on WinPython issue. You can see [version of this on my dropbox here](file:///C:/Users/hamis_000/Dropbox/Public/share2/dronekit-python/html/guide/getting_started.html#installing-dronekit-on-windows). Creating PR now.

djnugent, 2015-06-11:
Perfect

hamishwillee, 2015-06-26:
Quick start now removed

closed 2015-06-26 by hamishwillee