dronekit/dronekit-python #159: Best way to exit a script if you get into trouble?

hamishwillee, 2015-06-12:
If my battery goes low, or if a failsafe occurs, then it is likely the vehicle will RTL. In this case my script may well still be running. What is the best method to call to exit it/how can I exit it to MAVProxy without killing mavproxy?

djnugent, 2015-06-12:
```
while not api.exit and vehicle.mode.name != 'RTL:
      <your code here>
```
Your script ends when there is nothing else to execute. Just put everything in a while loop and provide a exit condition.

To kill the script from mavproxy run `api stop <thread #>`

djnugent, 2015-06-12:
Or for an immediate less clean method you could do this:

```
import sys

<your code before>

if vehicle.mode.name == 'RTL':
     sys.exit()

<your code that wont run after the script is killed>
```

hamishwillee, 2015-06-12:
@djnugent Thank you very much! The real example I'm looking at is that if my code is running and we go low battery there is an automatic RTL. However my goto code will actually reset the mode to guided and there can be some fighting over control in this case.

The first example isn't really all that useful because the loop is only checked when you enter the block. If you're doing lots of things in the loop then it simply isn't responsive enough. Of course you can use this around every operation, but then you end up having these loops all over the place. Yuk. 

The second looks more useful. I can add a vehicle mode callback and shut the script down from that. I'll try it :-)

As an aside ... we probably need a document on "good coding practices for drone scripts". If we were to write something like that, any other things you would mention? First thing that springs to mind is basic safety - don't do horizontal movement if your vehicle is at person height.

djnugent, 2015-06-13:
Yeah it really depends on how your script is structured. 
+1 on the good coding practices. 
Explain why `while not api.exit:` is important. 
Explain how to check for command completion for various commands. I love "simple goto" with the arm_takeoff function. That is the proper way to use dronekit and the docs don't explain that right now.

hamishwillee, 2015-06-13:
> That is the proper way to use dronekit and the docs don't explain that right now.

Working on it :-)  Coding practices would be a good place for this information.

IMO ``while not api.exit:`` is another case where you'd probably be better off having a callback ... if it were observable ... because the API can in theory exit at any time. Do you think we can make it observable?

djnugent, 2015-06-13:
@hamishwillee I'm sure we could make it observable. I'm not sure how "observe" works(a separate thread??) but it would be nice if the user didn't have to worry about it.

hamishwillee, 2015-06-13:
@djnugent This important enough to have its own topic and expand the interest - see #160. I don't know how observer works in this system, but certainly should be possible to provide some callback that people can register for and handle various failures how they like. We probably should create a separate issue. I haven't though.

hamishwillee, 2015-06-16:
@djnugent Just tried ``api stop 1``. As far as I can tell, this just sets the api.exit so it is up to your code to "do something" to exit. As before that's not really very useful unless most of your action is in a short event loop of some kind that gets checked regularly - we will need an observer of some kind for this to work in normal code.

I also tried sys exit. This does indeed stop the thread executing, but it doesn't clean it up. As you can see below the thread remains in the list after exiting:
GUIDED> API Threads:
  0: C:\Users\hamis_000\Documents\vagranttesting\dronekit-python\examples\guided_set_speed_yaw\guided_set_speed_yaw.py

Is there a way in code to completely exit the current thread of execution and clean up?

djnugent, 2015-06-16:
Yep they have their trade offs. I briefly looked into killing the thread and it is over my head. It appears the way api.exit is implemented is the standard way to kill a thread. I think if we want more control over threads/processes, mavproxy needs to implement multiprocessing instead on threading. This may be a question for @tridge

hamishwillee, 2015-06-17:
@djnugent I think the use of api.exit to notify that the thread should kill itself is fine, as is the use of sys.exit() for the thread to do the suicide :-) What we need to make this work is:

* Making the api.exit observable so that the app can kill itself at any time (rather than forcing effective code to write a tight event loop)
* Update the api.list() function so that it detects that the killed thread really is dead. If it isn't dead we need to look at why sys.exit() isn't killing it.

tridge, 2015-06-17:
Daniel, feel free to join mumble to discuss

hamishwillee, 2015-06-17:
Just a few notes from Tridge:
> there is a way, but not a neat one. self.mpstate.functions.process_stdin("module unload MYMODULENAME", immediate=True)
> We should add a better API. This function needs to be exposed to modules: https://github.com/Dronecode/MAVProxy/blob/master/MAVProxy/mavproxy.py#L279

@djnugent - see http://dev.ardupilot.com/wiki/ardupilot-mumble-server/ for information on Mumble setup.

djnugent, 2015-06-17:
@hamishwillee I talked with tridge and attempted to implement a solution. We came up with a solution for the script to terminate itself with a clean exit. However, it is not possible(based on my research) for a parent thread(mavproxy) to kill a child thread(your script). The script is the only thing that can terminate itself. That is why api.exit is in place. Mavproxy can't force the script to exit, but it can flag it to exit and have the script terminate itself. 

The reason a parent thread terminating child threads is not supported is because the parent thread is not aware of the child's activities. The child could be accessing a file or has an open connection to something. Terminating the child would cause an inconsistent state.
 
So...
Have the script terminate itself cleanly: Done
Mavproxy forcing the script to terminate immediately: Impossible(or extremely hard and dangerous)

mrpollo, 2015-06-17:
There's a case where event listeners on attributes would still get fired when we exit the thread, I think this is a bug and we should squash it before labeling this as "Done", but I agree this is the only clean way of going about it.

hamishwillee, 2015-06-18:
@djnugent  Thank you! So how do I test this (where do I get the latest version, and how do I call this new form of exit?)

Agree with @mrpollo previous comment that we need to make sure that exiting the script does clean up attribute observers etc.

Having a clean way for the thread to call exit is still only part of the solution. We also need to be able to observe the api.exit so that we don't need to write code that polls it (everywhere).

Even so, essential step in the right direction.

djnugent, 2015-06-18:
@hamishwillee I've made the changes on my local fork but something else took my attention before I could commit. Tomorrow.

@mrpollo how does the multiprocessing library sound instead of threading? That way we can terminate it if we please. 

An ideal situation would youd be able to throw an exception from the parent thread to the child and then have a 'final' block in the child thread to clean  up before exiting.

mrpollo, 2015-06-18:
@djnugent multiprocessing in mavproxy sounds like a nice addition you might want to talk to @tcr3dr who recently did some threading work on MAVProxy to fix some of the GUI components for OSX

hamishwillee, 2015-07-08:
Hi @tcr3dr (and @djnugent), can you please review this issue and also #166 - they appear to have got stuck!

Tim, the story is that #166 will allow a thread to cleanly kill itself - we definitely need that and it needs to be merged. 

The remaining problem here is that actually using the API is likely to be "messy" because you would have to litter the code with checks on api.exit. It would be better if api.exit was observable so you could kill the thread in a single callback. [Note that other cases you are likely to want to be able to kill the app are observable - e.g. if you detect RTL]

tcr3dr, 2015-07-21:
Due to progress on #188 I am more inclined to close this than to have it merged in for a short time. We might be looking at only a few weeks of utility, without a useful story for backwards compatibility.

@djnugent @hamishwillee what are your thoughts?

hamishwillee, 2015-07-21:
@tcr3dr This issue is a docs question that needs to be answered, not code that needs to be merged. It needs to be kept open and then answered when #188 delivers (or to put it another way, the requirement of this issue to be able to answer the question "what is the best way to exit a script if you get into trouble?" is something that #188 needs to answer *in a sensible way*.

I agree that #188 looks like a more complete solution to the problem than #166 and #167, so if @djnugent agrees, more than happy close those while we wait on the more complete solution.

hamishwillee, 2015-11-02:
Closing this - not relevant in DKPY2

closed 2015-11-02 by hamishwillee

tcr3dr, 2015-11-03:
(See #348)