dronekit/dronekit-python #160: Doc: Coding Best-Practice

hamishwillee, 2015-06-13:
Create a topic explaining coding best-practice for dronekit-python.

Should cover:
- Usage of api.exit
- What you should do if the connection fails (though no easy way to detect that yet)
- What you should do in script if forced into RTL
- Fact that this is a lossy connection - how to check for command completion for various commands.
- Similarly, explain why callbacks not guaranteed to complete (or if they are, how they work to do this in a lossy environment?)
- Safety issues - not moving horizontally until at safe height - any others?

@tridge @rmackay9 @CraigElder @mrpollo @billbonney @djnugent @tcr3dr  - Would appreciate your thoughts on this topic - what are the things that people need to consider when writing scripts for drones?

djnugent, 2015-06-24:
This isn't really coding practice, more linux knowledge, but the biggest issue I see people encounter is import errors. They are not importing the correct packages or modules when their script runs. Honestly I don't even know the proper solution. I just know that if I mess around with my PYTHONPATH, PATH, file permissions, and project structure long enough then it will work. I think we need to provide some do's and dont's when running scripts.

hamishwillee, 2015-06-24:
@djnugent I think that is definitely a good thing to include.

tcr3dr, 2015-11-03:
See #397.