dronekit/dronekit-python #163: DK Build Toolchain updates for Windows DroneKit Installer

hamishwillee, 2015-06-16:
Update our build system to automatically create new Windows DroneKit installation packages on every rebuild. This includes:

- Establish process for Windows versioning and distribution
- CI/CD builds for Windows on each release http://www.appveyor.com/

This needs to be addressed when #161 is merged.

tcr3dr, 2015-07-24:
This is actually doable now that Appveyor can run the installer. See #236.

I suppose we should publish builds on each tag release. This shouldn't be too hard to setup. @hamishwillee, should this happen before adding Windows Installer instructions to the tutorial?

hamishwillee, 2015-07-26:
@tcr3dr Yes please. I can't update the getting started to include the windows installer option until I know where the installers will be hosted (ideally with a permanent "latest version" link). That is part of what this issue needs to deliver.

tcr3dr, 2015-07-30:
@hamishwillee We have our first Windows builds:

* https://s3.amazonaws.com/dronekit-installers/windows/dronekit-windows-latest.exe
* https://s3.amazonaws.com/dronekit-installers/windows/dronekit-windows-v1.4.2.exe

These will continue to be built as we publish releases. Feel free to link to them!

tcr3dr, 2015-07-30:
@hamishwillee Feel free to close this issue in favor of #164.

hamishwillee, 2015-08-02:
Closed in favour of #164

closed 2015-08-02 by hamishwillee