dronekit/dronekit-python #164: Guide Updates to reflect Windows DroneKit Installer

hamishwillee, 2015-06-16:
Update the DK-Python docs to reflect the addition of new DroneKit installer for Windows. Topics to cover
- How to use it (where you get it, mapping of installer version to DK version, version of MavProxy required)
- When you use it - why use this and when use PIP package. 
- Using DroneKit vs / Working on DroneKit distinction in guides for Windows

This requires #163 be completed.

hamishwillee, 2015-06-16:
With respect to "When you use it - why use this and when use PIP package.". You use this as it is the simplest approach by far and it works! We haven't got a working "full" installation via the python command prompt yet so if you want the map and console and DK you need this. (in fact for SITL only you don't since SITL will show you the map and console on windows - but if you connect from windows to anything else then you do).

@stephendade, are there any limitations as compared to installing DK/Mavproxy in a command prompt? 

My understanding is that there aren't. The only potential limitation was that you might want to use a library not currently loaded by DK/MAVProxy - however there is nothing to stop you going to Python command prompt at C:/Python2.7/ and installing whatever librarys you need right?

hamishwillee, 2015-06-16:
@mrpollo Re 

> Using DroneKit vs / Working on DroneKit distinction in guides for Windows

We have a clear distinction on this already based on the titles of the Building vs Installing sections and the text explaining that [Building DroneKit-Python on Windows](http://python.dronekit.io/contributing/developer_setup_windows.html) is for developing.

I am assuming here that we don't change our current [Building DroneKit-Python on Windows](http://python.dronekit.io/contributing/developer_setup_windows.html) to use some process whereby they rebuild the installer and then reapply it. That would be more complex and have additional dependencies.

stephendade, 2015-06-16:
@hamishwillee 
>  are there any limitations as compared to installing DK/Mavproxy in a command prompt?

There are no differences that I am aware of. They should run exactly the same.

>however there is nothing to stop you going to Python command prompt at C:/Python2.7/ and installing whatever librarys you need right

The install is completely independent of whatever Python environment you've got, so that won't work. If the user's custom modules require other python libraries, I'm not sure how they'd do that - maybe if those libraries were installed with their module in the MAVproxy dir.

hamishwillee, 2015-06-16:
> The install is completely independent of whatever Python environment you've got, so that won't work. If the user's custom modules require other python libraries, I'm not sure how they'd do that - maybe if those libraries were installed with their module in the MAVproxy dir.

I don't quite get that - there was a stated dependency/assumption on C:/Python2.7/ so isn't that the "Python environment you've got"? i.e. if you open the command prompt and add something from PIP wont that be useable.

stephendade, 2015-06-16:
During the build process for MAVProxy (py -> exe) for the Windows build, it takes all the dependent Python libraries (and Python itself) and bundles it all up into the exe and associated support files. Then it take all this and puts it into a setup.exe.

Thus MAVProxy (and all modules running within it) are using a "mini-Python" environment contained within the MAVProxy.exe and associated files in the install folder.

stephendade, 2015-06-16:
The dependency on C:/Python2.7/ is for the initial build/compilation only.

hamishwillee, 2015-06-16:
Thank you - that makes perfect sense. I guess you might well set things up using the command prompt in order to get more control over these dependencies (rather than using the installer).

tcr3dr, 2015-07-30:
See these URLs for our first Windows builds:

* https://s3.amazonaws.com/dronekit-installers/windows/dronekit-windows-latest.exe
* https://s3.amazonaws.com/dronekit-installers/windows/dronekit-windows-v1.4.2.exe

Thanks again @stephendade for the work behind these.

hamishwillee, 2015-08-02:
@tcr3dr Can the parent directory ( https://s3.amazonaws.com/dronekit-installers/windows/ ) be given unrestricted *read* access - at the moment it is not possible to see the contents. 

That way I can say something like "get the latest here, but see dir HERE for legacy versions".

hamishwillee, 2015-08-03:
Docs issues are addressed by #268

jamesbellaero, 2015-08-15:
Both of the links stopped working.

hamishwillee, 2015-08-15:
@jamesbellaero Yes, but the ones on the [live site getting started](http://python.dronekit.io/guide/getting_started.html) do work - we pushed fixes for that today.

closed 2015-08-15 by hamishwillee