dronekit/dronekit-python #169: API to get information back from specific commands

hamishwillee, 2015-06-18:
At the moment we have attributes and some methods that let us query specific information from the vehicle (home position, location, gps etc). We can send MAVLink messages that "do" something - set Yaw, move to position, set servo value, etc. 

What we don't have is a public method to get arbitrary information back from the vehicle. So for example I can sent a message to the API with the message ID requesting the camera status or the current value of a particular servo, but I can't then get the response back into my application.

In particular it would be great to be able to query the capabilities of a particular target platform.

csmiler, 2015-06-25:
Hi Hamish,

Thanks for updating the project! I'm new to dronekit..Initially I was trying to read altitude via "vehicle.location.alt" every 0.1 sec, but it popped out 6 same readings although the drone was climbing up fast. At what rate do the vehicle attributes get updated? Alternatively I could "ask" the drone for altitude readings from on-board sensors (ultrasonic sensor for example), but your thread suggests that retrieving message from vehicle has not been implemented yet (which kinda explains why I didn't find any example getting messages from vehicle..) Any updates on getting info back from the vehicle, sensor readings etc.? 

Cheers,

csmiler

hamishwillee, 2015-06-26:
@csmiler This is more an autopilot question than a DroneKit question, and will be impacted by the autopilot implementation, hardware and the communication link (if you're over USB it will be potentially a lot faster than over an RF Telemetry link). DroneKit does process every message it receives. AFAIK there is nothing you can do about getting a faster rate than the comms link and hardware allow.

All that said, [this post](http://ardupilot.com/forum/viewtopic.php?f=64&t=10334) indicates someone was getting updates over USB around every 0.12seconds.

We still don't have a mechanism for getting "arbitrary" specific messages back from the system. If I "had" to do this I'd look at [set_mavlink_callback()](http://python.dronekit.io/automodule.html#droneapi.lib.Vehicle.set_mavlink_callback).

eliao, 2015-06-26:
@csmiler - check your stream parameters.  Especially SR0_Position:  http://copter.ardupilot.com/wiki/configuration/arducopter-parameters/#position_stream_rate_to_ground_station_sr0_position
The max stream rate is 10 hz, but yours may be set to a smaller number.  Many GCSes will set the stream rates to numbers smaller than 10 hz.

hamishwillee, 2015-06-26:
@eliao That's awesome - [the wiki](http://copter.ardupilot.com/wiki/common-optional-hardware/common-telemetry-landingpage/common-3dr-radio-advanced-configuration-and-technical-information/#mavlink_reporting) indicates that the firmware can use this information to automatically adapt the telemetry stream rates to the data rate that the radios can sustain - so does that mean the parameters just set the "maximum value"?

My gut feeling is that adding information on stream rates to DroneKit docs may not be appropriate, but perhaps we can do more on the wiki?

eliao, 2015-06-26:
I'm not sure what happens if you set the stream rates to something that your radios cannot sustain.  I assume it will start dropping messages, but I am not sure which messages get dropped.

csmiler, 2015-06-26:
@hamishwillee Thanks for the reply! (didn't expect to hear back from you so soon!) I understand there's physical limit over RF link. I'll try set_mavlink_callback() tomorrow. In fact I was looking at the function earlier today, but didn't go with it because of the conservative warning..on the other hand, I use channel_override..

@eliao That's a good point! I'll check my param. Thanks for the suggestion. I noticed that the telemetry log has much fewer data points (< 3 Hz) compared to dataFlash log which records at 10 Hz. Maybe the stream rates was too low.

hamishwillee, 2015-06-26:
@csmiler Please don't use ``channel_override()`` - @rmackay9 will hunt you down :-) I've just added a bunch of documentation on the right way to do [takeoff](http://python.dronekit.io/guide/taking_off.html), [position control](http://python.dronekit.io/guide/copter/guided_mode.html#position-control) and [velocity control](http://python.dronekit.io/guide/copter/guided_mode.html#velocity-control) which are recommended. While you're there, note the lovely new addition of the page table of contents in the left-sidebar :-)

csmiler, 2015-06-26:
@eliao I checked my parameters today. SR0_POSITION was 3 Hz! But even if I changed it to 10 Hz, the stream rate seems to be the same ( 2 or 3 points per sec)..I also played with other params, but couldn't increase the rate..below is the params I am using. I turned off SR1_xxx (telemetry 2 on Pixhawk in my 3DR IRIS+ is connected to FrSky receiver) [oddly everytime I reconnect to IRIS via mavproxy SR1_xxx params change to 4 except SR1_PARAMS 10, so I had to manually set them to 0 again]... Do you think there is anything else I can do to increase the stream rate, or maybe it's capped due to transmission limit (3 Hz is kinda low though..)? Thanks so much for your help!

SR0_EXTRA1       10.000000
SR0_EXTRA2       10.000000
SR0_EXTRA3       2.000000
SR0_EXT_STAT     2.000000
SR0_PARAMS       10.000000
SR0_POSITION     10.000000
SR0_RAW_CTRL     4.000000
SR0_RAW_SENS     2.000000
SR0_RC_CHAN      2.000000
SR1_EXTRA1       0.000000
SR1_EXTRA2       0.000000
SR1_EXTRA3       0.000000
SR1_EXT_STAT     0.000000
SR1_PARAMS       0.000000
SR1_POSITION     0.000000
SR1_RAW_CTRL     0.000000
SR1_RAW_SENS     0.000000
SR1_RC_CHAN      0.000000
SR2_EXTRA1       0.000000
SR2_EXTRA2       0.000000
SR2_EXTRA3       0.000000
SR2_EXT_STAT     0.000000
SR2_PARAMS       0.000000
SR2_POSITION     0.000000
SR2_RAW_CTRL     0.000000
SR2_RAW_SENS     0.000000
SR2_RC_CHAN      0.000000

eliao, 2015-06-26:
Are you using MAVProxy?  There's some logic in MAVProxy that automatically sets all the stream rates to 4 hz every few seconds.  You can pass it the "--streamrate=-1" option to disable this.

csmiler, 2015-06-26:
@eliao Yes I use MAVProxy. SR1_xx are set to 4 hz every time I reconnect via mavproxy, but after I manually set them to 0, they won't be reset to 4 after a few seconds, unless I disconnect and reconnect. But you are right about "--streamrate=-1"!! After I add the option, even if I reconnect the SR1_xx remains 0 except SR1_PARAMS automatically set to 10..Thanks!

The stream rate was the same though...Any idea would be appreciated~ :+1:

eliao, 2015-06-26:
@csmiler - You've exhausted my knowledge.   Maybe the link is maxed out?  You might be able to turn off some other streams to see if that helps.

csmiler, 2015-06-26:
@hamishwillee Yes I see you have added so much more stuff which makes it more intuitive and helpful for newbies like me :). The cute table on the left is a huge plus! Good work! :100: 

If I understand correctly, at the moment the only way to get info off the copter is to enable stream, and there is no such thing as "one-time talk" by sending one message asking for a current reading of one sensor and receiving the response message promptly which contains the info requested, am I right? So I cannot really "ask" for altitude every 0.1 sec (even if transmission is fast enough) and "receive" an answer, but can only take whatever is fed from the attribute "location.alt" which comes from VFR_HUD.alt. I checked the flood from set_mavlink_callback(), and messages for "ATTITUDE" and "VFR_HUD" are at ~ 2 hz, which is consistent with the limited stream rate..no luck there..

I learnt the functions "takeoff, position control and velocity control" earlier from your demo (thanks!!!). What I am trying to do here is to fly IRIS indoor (no GPS) with additional positioning system feeding my program with the copter's 3D location, and be able to fly to any coordinate within a say 2 m x 1 m x 1 m area within ONE second. The functions in GUIDED mode are not yummy since GPS is not locked. So the "naive" idea I had is to "mimic" a joystick by overriding channel 1~4 and write external PID controllers to provide channel values based on its current position and target position fed by additional positioning system (silly, right?). So all controls are off Pixhawk, and the only task for Pixhawk is to maintain stabilized...I am using "Stabilize" mode since it is least damped and most responsive to channel inputs given the one-second limit. Then it is crucial to "know" its position in a timely manner. Right now I'm just testing my PID controller using feedback from "vehicle.location.alt", but it turns out that the feedback is too slow because of the stream rate...any suggestion on indoor navigation problem Expert Willee?

Sorry if I am asking silly questions...I just started flying drones..not an easy thing but intersting!!

csmiler, 2015-06-26:
@eliao You've helped me a lot! Many thanks! It'll be unfortunate if the telemetry can only provide this much of stream rate...I'm getting a ultrasonic sensor for altitude, maybe I'd better hook it up to a raspberry pi to get its reading rather than connect to Pixhawk and stream it..

hamishwillee, 2015-07-07:
@csmiler Sorry, just got back from holiday. Yes, [your understanding above](https://github.com/diydrones/dronekit-python/issues/169#issuecomment-115912930) is correct. Unfortunately you've hit the limit of my knowledge here too - no more ideas :-(

hamishwillee, 2015-07-07:
@mrpollo Looks like some stuff in the "right direction" happening in Android for this: https://github.com/DroidPlanner/DroneKit-Android/pull/188

hamishwillee, 2015-11-19:
@tcr3dr I think we can do this now, simply by defining a message listener right?

tcr3dr, 2015-11-19:
@hamishwillee That's correct. In the constructor of a Vehicle, that's effectively how we track state changes anyway.

closed 2015-11-19 by tcr3dr

amilcarlucas, 2018-06-11:
The bandwidth can be determined by using the https://github.com/ArduPilot/pymavlink/blob/master/tools/mavtelemetry_datarates.py script