dronekit/dronekit-python #185: Continuous integration requirements

tcr3dr, 2015-07-08:
We want it! We need it!

dk-python has grown and our list of needs that need to be tested on merge has also. In this issue I want to compile a canonical list for CI that needs to be implemented. We can then split into separate issues talking about the technical details (when we use SITL, etc).

* [ ] Installation of dronekit-python from requirements.txt
* [ ] Installation of dronekit-python from setup.py
* [ ] `tests/` directory
* [ ] `examples/` directory
* [ ] Windows installer
* [ ] Tests / examples provided in documentation

tcr3dr, 2015-07-08:
Right now we are always testing against mavproxy "master", is there any reason this should be changed?

mrpollo, 2015-07-08:
also we need to write unit tests, and possibly integration tests running with SITL

hamishwillee, 2015-07-08:
> Right now we are always testing against mavproxy "master", is there any reason this should be changed?

No idea. One thing though, the instructions for MAVProxy on Windows use the installer version of MAVProxy and an installer version of DroneKit that can go on top of it - so slightly different than just using command line python.