dronekit/dronekit-python #188: DKPY as a standalone Python module

tcr3dr, 2015-07-09:
dronekit-python 1.0 is structured as a MAVProxy module. This had the following implications:

* Leverages MAVProxy's routing capabilities (serial/udp/tcp)
* Allows reuse of MAVProxy's object and state code
* Runs as a component of MAVProxy's threading model

Given the successful first release, we're increasingly running into this friction:

* Naive use of dronekit-python, expecting it to be single-threaded, creates race conditions. (https://github.com/diydrones/dronekit-python/issues/12, https://github.com/diydrones/dronekit-python/issues/74)
* Script lifetime management expects processes, not threads. ([#159](https://github.com/diydrones/dronekit-python/issues/159))
* MAVProxy contains several dependencies (GUI and numerical related) which may not be used by dronekit-python (https://github.com/Dronecode/MAVProxy/issues/159)

There are now strong reasons to consider moving dronekit-python to run as a "top-level" library:

* We avoid further complicating MAVProxy's process model by duplicating OS-level functionality.
* We allow reuse of dronekit-python as the "object layer" that we expect it to be.
* Mavlink routing becomes simpler, as a dronekit-python script becomes just another Mavlink endpoint.
* We improve automated testing by allowing dronekit-python to be tested in isolation.

Having outlined the reasons, we're looking for discussion on:

* How high-priority this is (what features are blocked by not having this implemented)
* What risks we need to mitigate (duplication of code, etc.)
* Outstanding technical questions

This is currently slated for the next major release of dronekit-python (v2.0.0).

tcr3dr, 2015-07-09:
One thing to note is that this is effectively blocked by #185. We want a capable test suite to prevent regressions from this transition.

eliao, 2015-07-09:
I look forward to this day!

mrpollo, 2015-07-09:
:+1: 

Another dependency we might want to look into either help with performance of replace is pymavlink https://github.com/mavlink/mavlink/tree/master/pymavlink

hamishwillee, 2015-07-09:
A bit pedantic, but "Naive use of dronekit-python" implies that the *user* can use DK in a less naive way to fix the problem. Perhaps "naive implementation" would be better.

Another feature I see this helping with is debugging. You could start of debugging sessions (and would no longer need to instrument code to start a session). This should make it easier to integrate with standard IDEs.

I think it might help with making it easier to write code to communicate with a script - passing in command line arguments and having the script wait on user input.

tcr3dr, 2015-07-17:
Using pymavlink as a dependency directly is now being explored in the https://github.com/diydrones/dronekit-python/tree/major branch.

tcr3dr, 2015-10-14:
We've gone ahead with this! Merged into master.

closed 2015-10-14 by tcr3dr

hamishwillee, 2015-10-14:
Anyone reading this, note that we use "master" as the bleeding edge development branch. While the change has been merged, at time of writing  DKPY2 (without mavproxy dependency) is in beta, and the published documents still reflect DKPY1.