dronekit/dronekit-python #195: How to control the vehicle w.r.t its own postion. For example if i want to move the vehicle forward w.r.t it self how would i do it?

ghoshabhra, 2015-07-10:


ghoshabhra, 2015-07-10:
if any one can tell me how to get the current yaw angle i can write a function to convert the relative x and y direction to N.E.D.

hamishwillee, 2015-07-11:
In an ideal world ArduPilot would provide the SET_POSITION_TARGET_LOCAL_NED relative to the body frame and you would have this out of the box.

But since it doesn't ... you can get the bearing and GPS location relative to current position and a meters offset using [these functions in the guide](http://python.dronekit.io/guide/copter/guided_mode.html#frame-conversion-functions). The [guided mode example](http://python.dronekit.io/examples/guided-set-speed-yaw-demo.html) provides a concrete demonstration.

Sufficient?

ghoshabhra, 2015-07-11:
what does the bearing function do

hamishwillee, 2015-07-12:
Hi 

>what does the bearing function do 

Allows you to calculate a bearing from one Location to another. This is in the link I pointed to.

Re reading your questions:

> For example if i want to move the vehicle forward w.r.t it self how would i do it?

This is shown in the example code I linked above - basically the Vehicle.commands.goto() method allows you to move with respect to the global frame. Since you can get the current position using Vehicle.location you can convert this into a position relative to the vehicle. The example function "get_location_metres" makes this easy.

> if any one can tell me how to get the current yaw angle i can write a function to convert the relative x and y direction to N.E.D.

The current yaw can be obtained using [Vehicle.attitude](http://python.dronekit.io/automodule.html?highlight=attitude#droneapi.lib.Vehicle.attitude). You can find this sort of thing out [using search](http://python.dronekit.io/search.html?q=yaw&check_keywords=yes&area=default).