dronekit/dronekit-python #198: Update Simulation section to use dronekit-sitl-runner

hamishwillee, 2015-07-13:
As introduced in #176 by @tcr3dr:

> At this point I'm comfortable floating [dronekit-sitl-runner](https://github.com/tcr3dr/dronekit-sitl-runner) as a cross-platform alternative to the Vagrant sections.

> This uses precompiled binaries on S3 that are generated in a separate CI step: https://github.com/tcr3dr/dronekit-sitl-builder Three versions (3.2, 3.3-rc5, and 3.4-dev) are supported and built for Windows, OS X, and Linux.

> Basically, we should just outline what assurances are needed from that tool to use as a replacement for the Vagrant install (which I assume is less desirable in all circumstances).

hamishwillee, 2015-07-13:
@tcr3dr @mrpollo Installation of the SITL binaries works on Ubuntu linux (in VM) and Windows

1.  Can you arrange for someone to independently test this on Mac OS X - Windows did not work [initially](https://github.com/3drobotics/dronekit-sitl-runner/pull/15), so we can't make any assumptions that Mac will work.

One "problem" is that the second part of using this solution is to call mavproxy to forward the SITL messages from the standard SITL TCP port to some localhost UDP ports. This requires that you install mavproxy too, which is a pain on Windows. I'd quite like to resolve that issue to make the simplest possible "complete" SITL installation before we roll this out.

tcauchois, 2015-07-13:
I'm on OSX, use MacPorts for my python 2.7 install.  pip install ... worked fine.  To run dronekit-sitl I end up invoking `python -m dronekit_sitl_runner.__init__` which isn't the prettiest (I need to invoke mavproxy similarly through `python -m MAVProxy.mavproxy`, but at least it looks nicer).  The -m flag won't let me name a function to call, so I had to add the following to __init__.py:

```
if __name__ == "__main__":
    main()
```

After that, `python -m dronekit_sitl_runner.__init__ --list` works as expected.  Running it and connecting MAVProxy, I saw a panic about the barometer not reading data and then the link went down.

tcr3dr, 2015-07-13:
Currently blocked by needing to re-run some builds. See https://github.com/3drobotics/dronekit-sitl-runner/issues/22

The issue @tcauchois posted is separate, I have to investigate why this is. I'm fine if for more complicated paths setups (macports pip, etc.) you need to run `python -m dronekit_sitl`, but right now that doesn't work. I consider that a defect.

hamishwillee, 2015-07-13:
@tcr3dr Thank you. I created https://github.com/3drobotics/dronekit-sitl-runner/issues/24 for the bug (that python -m is required)