dronekit/dronekit-python #201: next=___ and flush() doesn't change the current waypoint

a7scenario, 2015-07-14:
I'm running an APMcopter SITL with DroneKit/MAVproxy.  When DroneKit is "flying" in AUTO mode in the SITL with an 8 waypoint mission, the following function calls returns fine, but fail to actually change the current waypoint.  I've confirmed that the next(value) property setter does get invoked, it just doesn't actually change the current waypoint onboard the copter nor does it change what the next() getter returns (it was waypoint #1 before and waypoint #1 after the below function calls).  

            MyQuad.vehicle.commands.next=5
            MyQuad.vehicle.flush()

If I do a set_mode command after changing the value of next, the current waypoint will indeed change to #5 as intended.  Welcome any thoughts :)

hamishwillee, 2015-07-15:
G'day! You don't need to call flush after calling next. 

I've created [this gist test code](https://gist.github.com/hamishwillee/b72845955057adda189e) which creates 10 waypoints and then sets mode to AUTO. After changing to auto mode a for loop runs showing a 1 second loop value  and the current command number e.g. ``"Current command number: 2`` from vehicle.commands.next. When it gets to a command number of 5 it sets the number back to 2. If you wait a bit you will see that it eventually does reach command 3 and progress (this takes a bit because the copter has to travel a fair way to number 2 waypoint from 5.

I think therefore this is invalid. Can you please confirm and get back to me. If you disagree, please provide an example I can run.

hamishwillee, 2015-07-17:
I am closing this. Please reopen if you don't think the answer was sufficient.

closed 2015-07-17 by hamishwillee

a7scenario, 2015-07-20:
Similar to your test code, I am able to change the current waypoint if I haven't changed my mission plan mid-flight.  If I've recently overwritten the mission plan, I can't change the current waypoint unless I do a mode change (even if I'm already in AUTO).  Can you run this quick and dirty test in your SITL to see if it succeeds/fails to change the current waypoint?  
https://gist.github.com/a7scenario/85d62dffe7f2aba69161

hamishwillee, 2015-07-21:
According to the expert you should not need to change the mode from AUTO. What is most likely happening is that Vehicle.commands.next does not change the waypoint until it has completed the current waypoint - and any additional next commands you send in the meantime are dumped. If you need to change the target immediately I think you might try sending https://pixhawk.ethz.ch/mavlink/#MISSION_SET_CURRENT 

I think that sounds like a bug, so I'll ask for further review.

Also, I'm not sure your code is "rock solid" you might want to print out the waypoints that have actually been set at the end of creating the mission - since if these aren't being uploaded the problem might be that you can't set a next of 4 if you only have 1 waypoint.

In addition, you shouldn't set the home waypoint directly as you have done in the gist. Provided that you properly cleared the previous waypoints, the home position should be set automatically for you (there is a bug in this area, see #132 but you're not running into it.)

hamishwillee, 2015-07-21:
Let me know how your testing goes (confirm above please!)

a7scenario, 2015-07-21:
Thanks for the thorough response.  I'll try to respond to each point:

--Good idea, but I actually already tried manually changing the current waypoint by using DroneKit's message_factory function mission_set_current_encode() and then using send_mavlink().  It exhibits the same behavior as when I use the next() property setter -- that is, it doesn't change the current waypoint under the conditions I described above.  Unfortunately, I need to change the current waypoint immediately and can't wait for AUTO to finish its active waypoint first.  

-The code is def not rock solid -- it's just something I ginned up to demonstrate the odd behavior.  That said, I'm confident that the waypoints are getting written as I'm running a GCS and SITL in parallel and can visually see the drone fly the 8 waypoints and I can download them off the simulated Pixhawk on demand.  

-Thanks for pointing this out!  I didn't realize that clear() doesn't clear out waypoint 0.

hamishwillee, 2015-07-22:
Hi

FYI, apparently the behaviour not quite as I indicated above. The next call is immediate. However if you load a new mission then the current target will complete first before the new position command is loaded.

Also discovered this bug today: https://github.com/diydrones/dronekit-python/issues/227

Cheers
H

hamishwillee, 2015-07-22:
OK, you can see some test code here https://gist.github.com/hamishwillee/40f51c05607834ded821#file-gistfile1-txt-L274 . It is a bit complicated, but it does appear to show code behaving "properly".

Ie if you change next then it changes the target waypoint. If you clear and load a new mission then you continue moving towards the same waypoint number, but this is a new waypoint. 

The action mostly happens in the `waitinfo(19)` functions which print out the current next, from this get the target waypoint, and print out the distance to that waypoint. After the first 19 seconds the code clears the commands and starts printing out info again. 

The log below (cut back) shows the command before and after clearing. YOu can see the "next" (Command num) stays the same, but the distance to the waypoint changes
```
 Command num: 10
 Time:  17
Location: Location:lat=-35.3605364,lon=149.1725917,alt=8.93999958038,is_relative=False
MISSION_ITEM {target_system : 255, target_component : 0, seq : 10, frame : 3, command : 16, current : 0, autocontinue : 1, param1 : 0.0, param2 : 0.0, param3 : 0.0, param4 : 0.0, x : -35.3605690002, y :
149.173782349, z : 30.0}
Distance WP: 132.592083203
Command num: 10
 Time:  18
Loading a new mission
CLEAR MISSION
Got MAVLink msg: MISSION_ACK {target_system : 255, target_component : 0, type : 1}
 Add a few new mission item to end then flushing
HERE IS JUST MISSION BEING UPDATED
APM: flight plan received
re-requesting WP 15
Location: Location:lat=-35.3605375,lon=149.1726044,alt=9.96000003815,is_relative=False
MISSION_ITEM {target_system : 1, target_component : 0, seq : 10, frame : 3, command : 16, current : 0, autocontinue : 0, param1 : 0, param2 : 0, param3 : 0, param4 : 0, x : -35.3604762, y : 149.175855099
, z : 30}
...
Distance WP: 361.930505167
Command num: 10
```

hamishwillee, 2015-07-23:
Also note this one - https://github.com/diydrones/ardupilot/issues/2584 - basically if the mission ends and you add new waypoints then the mission doesn't restart - it just waits. YOu can restart by toggling to guided and back to AUTO

a7scenario, 2015-07-23:
Thanks, I'll try out that test code.  And yes, per your second post, that is what I've been doing to hack around this issue -- toggling from AUTO to GUIDED to AUTO to force the next(value) command through.  In the SITL simulation, it's a little hard to tell if those quick mode changes have any impact beyond a little sputtering on the HUD.  The other workaround, I would think, is to just make note of your current waypoint number (e.g. waypoint 4) before you clear the mission and custom craft the mission plan you upload to the pixhawk so that your intended immediate destination is placed into that current waypoint "slot" (e.g. waypoint 4).  Just a thought.

hamishwillee, 2015-07-23:
There shouldn't be any real impact changing from AUTO to GUIDED to AUTO though it does depend on the parameter which defines what to do when you go into auto mod - you can either resume or restart from the beginning. The important thing when switching is to go to a mode where RC input is ignored (like guided) otherwise you might accidentally change throttle during the switch. 

Yes you can keep the current waypoint. I think an easier approach is to always just set one* nav waypoint (and as many condition waypoints as you need first) and then clear and upload the next one. Then you've just recreated guided mode.

(*) The only problem is that you can't monitor ``next`` because it will always be "0". The workaround I am playing with here is uploading 2 waypoints. The second is loiter_unlim, so you can tell when you reached your destination. Also just a thought !